256dpi/arduino-mqtt #2: Compiling error

mohitjain1962, 2015-04-11:
I am getting an error message - 'MQTTClient' does not name a type - unable to proceed

256dpi, 2015-04-12:
- Did you install the library manually or via the new Library Manager?
- Did you include the library in your sketch?

Maybe you could in general append your sketch and some more context?

mohitjain1962, 2015-04-12:
Dear Joel,

Thank you for taking out the time to reply.

1. I installed the library manually.

2. Yes, I included the library in my sketch.

3. I got errors related to “callback”

Regards
Mohit

256dpi, 2015-04-12:
As an "error" could possibly be anything, do you maybe have some more details?
Eg. a screenshot or your code?

256dpi, 2015-04-13:
I guess you found the problem on your own. I will close the issue off for now.

closed 2015-04-13 by 256dpi