256dpi/arduino-mqtt #13: TLS Support

ekarlso, 2015-12-03:
Hi, any plans on including TLS or showing a TLS support example?

256dpi, 2015-12-03:
Hi @ekarlso

I haven't tried TLS yet. For sure, it would be pointless to try with a normal Arduino Uno and the standard Wifi or Ethernet Shield, as the processor is too slow and constrained. But the Arduino Yun should support it as the network is handled by the Linux processor and there is also the new 101 Wifi Shield that supposedly has an integrated encryption chip.

What hardware do you got?

noelgeorgi, 2016-03-21:
@ekarlso @256dpi TLS works just use **WiFiSecureClient** instead of **WifiClient** with ESP8266 and its works ok

256dpi, 2016-03-21:
@noelgeorgi Cool! Thanks for the report. Could you send me and example like [these](https://github.com/256dpi/arduino-mqtt/tree/master/examples) that I can add?

noelgeorgi, 2016-03-21:
added a pull request

gibix, 2016-03-22:
The tls works perfectly also on Zero+Wifi101 and MKR1000, the sketch in quite the same of the ESP, just:
```
#include <SPI.h>
#include <WiFi101.h>
#include <WiFiSSLClient.h>

WiFiSSLClient net;
```

256dpi, 2016-03-22:
@gibix Thanks for reporting on these additional devices as well! Would you mind adding some examples for the Arduino Zero, Wifi101 and MKR1000? I currently do not have access to these devices...

gibix, 2016-03-22:
@256dpi added #26

256dpi, 2016-04-03:
Closing this as the PR is almost done.

closed 2016-04-03 by 256dpi