256dpi/arduino-mqtt #16: Ethernet/Wifi Shield support?

q2dg, 2015-12-13:
Is it feasible?
Thanks!

256dpi, 2015-12-14:
Hi @q2dg 

It should work out of the box, all tough I never tested it as I do not have the particular hardware at hand.

I added some examples in this PR: #17 that show how it would be done.
Could you try these and give feedback if they work?

q2dg, 2015-12-14:
Yes, they work perfectly!
Thanks a lot!!!

PS: Could you modify the README accordingly?

256dpi, 2015-12-14:
Yup I'm just about to do that. Just to be sure: Did you test both examples?
Are the connections stable for more than 5 minutes?

q2dg, 2015-12-14:
Ahh, I haven't tested more than 5 minutes, sorry. I'll try again

256dpi, 2015-12-14:
I think the WiFI example had a little bug, I updated the example.

q2dg, 2015-12-14:
Yes, you're true, I was going to tell you now: its behaviour is a little erratic

256dpi, 2015-12-14:
Could you be a little bit more specific about what does work and what not?
Otherwise it's a little hard to help from my side, and I would like to merge #17 soon.

Thanks for your help!

256dpi, 2015-12-15:
See #17.

closed 2015-12-15 by 256dpi

q2dg, 2015-12-15:
Sorry for answering so late...Yesterday I was waiting long time with my Wifi Shield executing example code to see if there was any problem and later I forgot. But today I've seen it still works well so...yes, with last changes it works well!
Thanks!!!

256dpi, 2015-12-16:
Cool! Thanks for the feedback!