256dpi/arduino-mqtt #19: Exceeding length crashes devices

karlp, 2015-12-29:
If I exceed the max length of the MQTT packet, I expect to get some sort of failure.  Instead, on ESP8266 at least, I get a system crash.  I can't see any way of doing this without changing the API unless you simply ignore overly long messages.

This is possibly a bug in paho embedded of course, I'm having a bit of a hard time finding who's entirely responsible.

256dpi, 2015-12-29:
Publishing too long messages should be safe as it is checked here: https://github.com/256dpi/arduino-mqtt/blob/master/src/lib/MQTTSerializePublish.c#L62. But receiving too long messages will crash the application as there are no checks implemented yet: https://github.com/256dpi/arduino-mqtt/blob/master/src/lib/MQTTDeserializePublish.c#L35. Sadly, this is a limitation of the paho embedded library.

I'll look into this but it might be complicated to resolve quickly.

256dpi, 2017-05-03:
This has been fixed in #59. The code will return false on `yield()` if the receive buffer is too small.

closed 2017-05-03 by 256dpi