256dpi/arduino-mqtt #21: cannot connect to test.mosquitto.org

marcobrianza, 2016-03-03:
Hi Jöel, thanks for the processing fix about bytes in the other thread.

While with processing test.mosquitto.org works perfectly, I cannot make it work with Arduino.
I have verified that broker.hivemq.com works both in Arduino and Processing.

my example with Arduino is this:
https://github.com/marcobrianza/wemakecolors/blob/master/Arduino/test_WiFiMQTT/test_WiFiMQTT.ino

Marco

256dpi, 2016-03-04:
Hi Marco, Before I look deeper into this, what kind of hardware are you using?

marcobrianza, 2016-03-04:
Hi I'm using ESP8266, board nodeMCU 1.0.
I have just setup my mosquitto server and the lib connects correctly. Anyway it will be good if also  the public server works.

256dpi, 2016-03-07:
I will do some test with hardware I have around and get back to you.

256dpi, 2016-04-01:
Until now the client sent empty strings as username and password if there were none specified. As that worked with other brokers it didn't with test.mosquitto.org. With the latest version the client should now properly format the connect packet.

@marcobrianza Once again, can you confirm?

256dpi, 2016-04-14:
I consider this fixed as well.

closed 2016-04-14 by 256dpi