256dpi/arduino-mqtt #22: client.connected() returns true after stopping mqtt server

marcobrianza, 2016-03-04:
Hi Jöel is still me testing hard your framework :)

I have noticed that client.connected() returns true after stopping the MQTT server.

on the ESP8266 sending a message with the client disconnected crashes the board.

256dpi, 2016-03-07:
As this library is just calling the Paho embedded client it might be a limitation of their code, or the protocol stack of the Arduino or any other platform. As with the other issue I will conduct some tests with my hardware.

256dpi, 2016-04-01:
With the latest version this should also work now.

256dpi, 2016-04-01:
@marcobrianza Could you test if the problem still exists with your setup or has been resolved?

marcobrianza, 2016-04-01:
ok, thank you I'll test it in the next days.

matlus, 2016-04-10:
@256dpi , I just tested is. In my loop() function I have code like this

```
  if (!mqttClient.connected())
  {
    connectToWifiAndMqttBroker();
  }
```
It never falls into the condition. That is the connectToWifiAndMqttBroker() function never gets called even when the broker is stopped. Wifi is still connected.

What really happens is that after some time the connect to the broker gets disconnected. But due to this issue, I can never reconnect. Similar code works on the Particle Photon and also when using the PubSubClient library.

256dpi, 2016-04-14:
I was finally able to fix this issue and successfully tested it with my hardware. The bug is related to a limitation in the underlying library. Please test again with the current version 1.9.6.

Reopen this issue if there are still some problems remaining.

closed 2016-04-14 by 256dpi

matlus, 2016-04-14:
@256dpi ,
Yes, this problem has been resolved with the update.
Thank you!