256dpi/arduino-mqtt #23: WDT reset when connection lost

noelgeorgi, 2016-03-21:
WDT reset occurs with ESP8266 sketch after connection lost from MQTT broker.. I this the normal behaviour??

256dpi, 2016-03-22:
Hi thanks for reporting this! My guess is that during a reconnect something nasty happens which will let the ESP crash while the Arduino would simply restart (problem solved ;)). Do you have any code at hand that illustrates the problem? ( A stripped down version might be enough)

noelgeorgi, 2016-03-22:
I used the AdaFruitHuzzahESP8266 sketch in the examples without any change. I tried disconnection by simply restarting the MQTT  broker in my server, which other libraries like PubSubClient and AdafruitMQTTLibrary was able to reconnect sucsessfully.

noelgeorgi, 2016-03-23:
@256dpi any progress?

256dpi, 2016-04-01:
Hi @noelgeorgi I released a new version that should fix this problem. The problem was that the network implementation didn't check if the client is still connected when writing to the stream.

Please give me an update if the fix helped!

matlus, 2016-04-01:
I tried reproducing this OPs scenario. I'm running mosquito locally so it was easy enough to do. Also, I have code in the loop method like so:

`
void loop() {
  if (!mqttClient.connected())
  {
    connectToWifiAndMqttBroker();
  }
  yield();
  mqttClient.loop();
}    
`
In the connectToWifiAndMqttBroker() method I've got various Serial.println() statements. However, I find that this method is not called when I stop the broker service. Restarting the service doesn't help either.

So, the question is - would the mqttClient.connected() method return false without calling disconnect() first?

256dpi, 2016-04-02:
Your guess is almost right. The problems is that the client only sets the internal boolean to false if one of the following method fails: publish() subscribe() and unsubscribe(). In that case, a client that is doing nothing would currently never realize that it is not connected anymore.

A fix would be that we check the status of the underlying stream on connected(), but I want to investigate first why a network error does not traverse up to the client when checking for new incoming bytes.

256dpi, 2016-04-14:
As #22 has been resolved this problem should also be fixed.

closed 2016-04-14 by 256dpi