256dpi/arduino-mqtt #28: WDT Soft Reset when Not using SSL (port 1883)

matlus, 2016-04-01:
Ok, this is really weird but I'm able to reproduce this....

I'm porting code from using the PubSubClient library to this library. As you can imagine I had a few incompatibility issues. Nonetheless, I pressed on and but I was foxed with this one issue and couldn't figure out what I was doing wrong. So eventually, I use one of the Example sketches in order to provide a minimum reproducible case.

`// This example uses an Adafruit Huzzah ESP8266
 // to connect to shiftr.io.
 //
 // You can check on your device after a successful
 // connection here: https://shiftr.io/try.
 //
 // by Joël Gähwiler
 // https://github.com/256dpi/arduino-mqtt

#include 
 #include 

const char *ssid = "ssis";
 const char *pass = "pass";

WiFiClientSecure net;
 MQTTClient client;

unsigned long lastMillis = 0;

void connect(); // <- predefine connect() for setup()

void setup() {
 Serial.begin(9600);
 WiFi.begin(ssid, pass);
 client.begin("iot.eclipse.org", 1883, net); // MQTT brokers usually use port 8883 for secure connections

connect();
 }

void connect() {
 Serial.print("checking wifi...");
 while (WiFi.status() != WL_CONNECTED) {
 Serial.print(".");
 delay(500);
 }

Serial.print("\nconnecting...");
 while (!client.connect("arduino")) {
 Serial.print(".");
 }

Serial.println("\nconnected!");

client.subscribe("/example");
 // client.unsubscribe("/example");
 }

void loop() {
 client.loop();
 delay(10); // <- fixes some issues with WiFi stability

if(!client.connected()) {
 connect();
 }

// publish a message roughly every second.
 if(millis() - lastMillis > 1000) {
 lastMillis = millis();
 client.publish("/hello", "world");
 }
 }

void messageReceived(String topic, String payload, char * bytes, unsigned int length) {
 Serial.print("incoming: ");
 Serial.print(topic);
 Serial.print(" - ");
 Serial.print(payload);
 Serial.println();
 }`

The changes I've made are:
 1. I'm not using SSL, so I've changed the port to 1883
 2. I'm using iot.eclispe.org as the MQTT broker
 3. I'm using the connect method's overload that takes just the clientId parameter.

Now I have other cases (different code) where when I use a clientId of "NestEyeESP8266", I get the same soft WDT reset. However, if I change the clientId to "NestEyeESP826" or "NestEyeESP82A66" everything works as expected. I don't mind sharing this code privately if needed.

256dpi, 2016-04-01:
Thanks for reporting this. I will look into this in the next couple of days!

matlus, 2016-04-01:
I just noticed that in the code here I'm using the WiFiClientSecure. However, the case I mentioned at the end of the earlier post (clientId="NestEyeESP8266") happens even with the non-secure version (WiFiClient).

256dpi, 2016-04-01:
First of all, you cannot use the SecureClient with a non secure port of a broker. That would raise an exception and on the ESP result in a "Soft WDT reset".

I tested "iot.eclipse.org" with both the normal and secure client as well both client ids you have supplied and the examples sketches work for me.

However, many MQTT brokers only allow one client using a particular clientId. If another client connects to "iot.eclipse.org" the connection to the client connected before will get closed immediately. In contrast, "shiftr.io" allows multiple clients with the same clientId. The "Soft WDT Reset" here is probably a bug and has been reported already here #23.

Did that help?

256dpi, 2016-04-01:
Update: I released a new version that fixed the bug which caused a crash on reconnecting.

matlus, 2016-04-01:
@256dpi  thanks. I'll see if this resolves my issue as well.

matlus, 2016-04-10:
Yes, this issue has been resolved.

closed 2016-04-10 by matlus