256dpi/arduino-mqtt #32: why distinguish mqttclient and yunmqttclient

gibix, 2016-05-03:
Is not better for the user's api to have only a single constructor for both mqttclient and yunmqttclient?
Considering the current code status could be quite easy to enable this feature at compile time with some `#if defined(__AVR_ATmega32U4__)`

If it is considered a good improvement I can make a PR.

256dpi, 2016-05-03:
I actually only use the YunMQTTClient if I need more program space. Also since the library targets now a bigger variety of boards I thought about removing that part to make the whole project a little simpler.

Do you think I makes sense to keep the code at all? I have the feeling you are anyway heading towards boards without a Linux part as with the Yun, correct?

gibix, 2016-05-04:
I like the idea to don't keep the library arduino-centric, maybe the yunclient could be a separated project that maybe use the mosquittolib that now is part of the yun-openwrt core and can be easly installed on others linux-like boards like artik.

256dpi, 2016-05-05:
I added #34 to the v2 milestone, as this would be a breaking change.

closed 2016-05-05 by 256dpi