256dpi/arduino-mqtt #34: Remove YunMQTTClient

256dpi, 2016-05-05:
As the library supports more and more boards and platforms like ESP8266 the specific `YunMQTTClient` implementation makes not that much sense anymore. The proposal is to remove the implementation and focus on lowering the memory footprint for the c++ implementation.

sandeepmistry, 2016-05-30:
Let me know if I can help out with this.

With the changes proposed in #42 the difference in free RAM is a bit more:
 * ArduinoYun_MQTTClient:
   * Free RAM: 1532 bytes
   * Code size: 18258 bytes
 * ArduinoYun_YunMQTTClient:
   * Free RAM: 1937 bytes
   * Code size: 14086 bytes

So there is a trade off in both code space and RAM when using a custom bridge.

256dpi, 2016-05-31:
Thanks for providing the measurements!

After thinking about this for a while I have the feeling we should keep the code in this repository. Also thanks to @sandeepmistry's PRs the implementation will be more up to date. The Library will always be Arduino-centric and therefore it should also try to provide the best available implementation per board and shield. Technically the YunMQTTClient could in the future also support QoS 1 and 2 as the linux side would offer the required persistence, process power and memory.

@lorenzoromagnoli, @gibix: Please submit your opinion as well!

closed 2016-07-08 by 256dpi