256dpi/arduino-mqtt #35: Implement support for Particle Photon(&Electron)

tlindener, 2016-05-06:
Hi,

it would be really great if you could implement support for the Particle Photon platform. 

Thanks a lot!

256dpi, 2016-05-10:
This might be a little complicated. As far as I know, the Particle API is similar to Arduino but does not use the whole Arduino Stack. @fabiantheblind Does the Particle have the same WiFiClient classes as found in the Arduino standard library?

fabiantheblind, 2016-05-10:
I don't think that they are compatible. Here are the docs for the Photon https://docs.particle.io/reference/firmware/photon/#wifi  

For me it worked to use https://github.com/rwaldron/johnny-five and then bypass the particle cloud and connect to shiftr.io or some own broker. See https://github.com/FH-Potsdam/bit/blob/master/src/index.js  

But it is all JavaScript.

256dpi, 2016-05-10:
@fabiantheblind thanks for the explanation!

@tlindener As we'd like to keep this library Arduino centric, you need to look at a projects like this: https://github.com/hirotakaster/MQTT.

closed 2016-05-10 by 256dpi