256dpi/arduino-mqtt #38: Change messageReceived signature?

256dpi, 2016-05-17:
I have the feeling the current signature is a little complicated for amateurs.

So rather than:

```c++
void messageReceived(String topic, String payload, char * bytes, unsigned int length) {
  Serial.print("incoming: ");
  Serial.print(topic);
  Serial.print(" - ");
  Serial.print(payload);
  Serial.println();
}
```

We could have something like this:

```c++
void messageReceived(MQTTMessage msg) {
  Serial.print("incoming: ");
  Serial.print(msg.topic());
  Serial.print(" - ");
  Serial.print(msg.payload());
  Serial.println();
}
```

Then we would also have the ability to add more convenience functions like:

```c++
if (msg.topicBeginsWith("/namespace")) {
  // do something
}
```

256dpi, 2016-05-17:
Actually this would also fix a serious bug as creating a String from the payload using `String(payload)` allocates another buffer with the exact copy.

gibix, 2016-05-18:
Is not very user friendly, but can be better!
You can add the sign if MQTTMessage in the readme or as comment in the examples?

256dpi, 2016-05-24:
What is not very user friendly? The current implementation or this proposed change?
Sorry, I read your answer now various times but still don't get it... 😄

This would be a breaking change and result in releasing a v2.

gibix, 2016-05-24:
Sorry for the language ;)
I'm saying that a lot's of arduino users may not be friendly with data structures and usually in the libraries are avoided, but if you don't want to hack the MQTTClient.h code I agree that use the MQTTMessage in the callback is the best way.
I suggest that in the readme we should add the structure description, like:

```
typedef struct {
    char * topic;
    char * payload;
    unsigned int length;
    boolean retained;
} MQTTMessage;
```

or in the examples as a comment.

256dpi, 2016-05-25:
Thanks for the clarifications.
I'm not yet fully convinced by my own idea, so I have to sleep on it again some time. ;)

I'll keep you updated.

sandeepmistry, 2017-01-26:
Regarding https://github.com/256dpi/arduino-mqtt/issues/38#issuecomment-231282819, in general it would be nice to avoid pointers in the call back, as from my understanding most Arduino users aren't exposed to them.

256dpi, 2017-05-03:
Has been simplified in #59.

closed 2017-05-03 by 256dpi