256dpi/arduino-mqtt #44: Undefined reference to Process::~Process()

Deric1111, 2016-06-15:
Hi,

I am writing a library on top of arduino-mqtt for internal use. However I am getting weird error like Process::~Process() has undefined reference. The error messages are as below:

> Arduino: 1.6.9 (Windows 7), Board: "Arduino Yún"
>
>libraries\FalconMQTT_Yun\FalconMQTT_Yun.cpp.o: In function `~YunMQTTClient':
>
>C:\Users\deric.dominic\Documents\Arduino\libraries\MQTT\src/YunMQTTClient.h:12: undefined reference to `Process::~Process()'
>
>libraries\MQTT\YunMQTTClient.cpp.o: In function `Process':
>
>C:\Program Files (x86)\Arduino\libraries\Bridge\src/Process.h:28: undefined reference to `vtable for Process'
>
>C:\Program Files (x86)\Arduino\libraries\Bridge\src/Process.h:28: undefined reference to `vtable for Process'
>
>C:\Program Files (x86)\Arduino\libraries\Bridge\src/Process.h:28: undefined reference to `vtable for Process'
>
>C:\Program Files (x86)\Arduino\libraries\Bridge\src/Process.h:28: undefined reference to `vtable for Process'
>
>libraries\MQTT\YunMQTTClient.cpp.o: In function `YunMQTTClient::updateBridge()':
>
>C:\Users\deric.dominic\Documents\Arduino\libraries\MQTT\src/YunMQTTClient.cpp:22: undefined reference to `Process::runShellCommand(String const&)'
>
>C:\Users\deric.dominic\Documents\Arduino\libraries\MQTT\src/YunMQTTClient.cpp:23: undefined reference to `Process::runShellCommand(String const&)'
>
>C:\Users\deric.dominic\Documents\Arduino\libraries\MQTT\src/YunMQTTClient.cpp:24: undefined reference to `Process::runShellCommand(String const&)'

>C:\Users\deric.dominic\Documents\Arduino\libraries\MQTT\src/YunMQTTClient.cpp:26: undefined reference to `Process::~Process()'
>
>libraries\MQTT\YunMQTTClient.cpp.o: In function `YunMQTTClient::connect(char const*, char const*, char const*)':

>C:\Users\deric.dominic\Documents\Arduino\libraries\MQTT\src/YunMQTTClient.cpp:43: undefined reference to `Process::begin(String const&)'
>
>C:\Users\deric.dominic\Documents\Arduino\libraries\MQTT\src/YunMQTTClient.cpp:44: undefined reference to `Process::addParameter(String const&)'
>
>C:\Users\deric.dominic\Documents\Arduino\libraries\MQTT\src/YunMQTTClient.cpp:45: undefined reference to `Process::addParameter(String const&)'

>C:\Users\deric.dominic\Documents\Arduino\libraries\MQTT\src/YunMQTTClient.cpp:46: undefined reference to `Process::runAsynchronously()'
>
>C:\Users\deric.dominic\Documents\Arduino\libraries\MQTT\src/YunMQTTClient.cpp:83: undefined reference to `Process::close()'

>libraries\MQTT\YunMQTTClient.cpp.o: In function `YunMQTTClient::publish(char const*, char*, unsigned int)':
>
>C:\Users\deric.dominic\Documents\Arduino\libraries\MQTT\src/YunMQTTClient.cpp:115: undefined reference to `Process::write(unsigned char)'
>
>libraries\MQTT\YunMQTTClient.cpp.o: In function `YunMQTTClient::connected()':
>
>C:\Users\deric.dominic\Documents\Arduino\libraries\MQTT\src/YunMQTTClient.cpp:172: undefined reference to `Process::running()'
>
>collect2.exe: error: ld returned 1 exit status
>
>exit status 1
>Error compiling for board Arduino Yún.
>
>This report would have more information with
>"Show verbose output during compilation"
>option enabled in File -> Preferences.

Any idea on this issue?

256dpi, 2016-06-15:
Wow, this is weird. Just to be sure: Did you select the Arduino Yun in the board menu before compiling?

256dpi, 2016-06-19:
Have you been able to solve the error?

closed 2016-07-08 by 256dpi