256dpi/arduino-mqtt #46: Wifi101 example not working

GoMapur, 2016-07-07:
Hello, I keep getting this when I basically copied the code and tried it on my Uno/Wifi101 Combination. 
checking wifi...
connecting.............
Seems like that it could not connect to the broker.shiftr.io
Does someone have the same issue?

Thank you!

sandeepmistry, 2016-07-07:
Hi @GoMapur,

Have you flashed the root certificate as per the comment at the top of the sketch? You'll need to do this for the "broker.shiftr.io" host.

https://github.com/256dpi/arduino-mqtt/blob/master/examples/ArduinoWiFi101_SSL/ArduinoWiFi101_SSL.ino#L6-L7

GoMapur, 2016-07-07:
Hi @sandeepmistry 
I used the plain one, unencrypted. Is SSL client required?

Thank you!

sandeepmistry, 2016-07-07:
> I used the plain one, unencrypted. Is SSL client required?

No, flashing the root certificate is not needed for unencrypted then.

GoMapur, 2016-07-07:
@sandeepmistry 
This problem traps me for two days, I worked around different MQTT libraries on Uno + Wifi101 but all of them failed at the connection stage. However, if I solely test Wifi connection it works perfectly fine. Do you have any idea what is going on?

GoMapur, 2016-07-07:
@sandeepmistry 
Hi there, I give up. For those who have the same issue, I suggest them use http://dweet.io/ to do the data logging and https://freeboard.io to finish instant data visualization. LOL

256dpi, 2016-07-08:
Hi @GoMapur, have you made any changes to the example?

sandeepmistry, 2016-07-08:
I just tried the `ArduinoWiFi101` example on my Uno and WiFi101 shield. It worked without any issues:
```
checking wifi...
connecting...
connected!
```

My WiFi101 shield has firmware version 19.4.4 running on it and I'm using version 0.9.1 of the the WiFi101 library.

256dpi, 2016-07-08:
@sandeepmistry Thanks a lot for testing this!

closed 2016-07-08 by 256dpi

nospam2k, 2019-12-05:
I think the problems is using ESP-01. I just tried this and get the same problem connecting as others.

256dpi, 2019-12-05:
@nospam2k Please update to 2.4.5. There has been a regression in the previous release.