256dpi/arduino-mqtt #47: esp8266 blocked

johnpica, 2016-07-07:
Hi 
ESP8266_SSL get blocked with this library when receives more than 4-5 messages in the same time. Messages are greater then 1000 bytes.

Ex:
60,1730,294,866,50,44,52,140,52,44,56,38,58,38,56,38,56,40,56,40,56,38,56,136,56,38,58,40,54,138,54,138,54,40,54,138,50,142,52,140,58,134,56,136,56,38,58,38,56,38,58,38,56,38,58,40,54,42,48,46,50,46,48,46,50,46,48,48,50,44,52,44,50,44,52,44,52,42,52,44,52,42,52,44,50,46,50,44,52,46,48,46,50,46,50,44,50,46,48,48,48,46,50,46,50,44,52,44,50,44,52,44,52,140,50,140,58,282,290,866,58,136,54,40,52,46,48,46,50,46,48,46,54,42,50,44,52,44,50,142,50,44,52,44,52,140,50,44,52,140,52,140,52,140,50,142,50,144,54,138,54,40,54,42,50,44,52,44,52,42,52,44,52,42,52,44,52,42,52,44,52,44,52,44,50,46,48,46,50,46,52,44,50,44,50,46,54,40,54,42,54,42,52,42,54,42,54,40,54,42,54,40,54,42,54,40,56,40,54,44,50,44,52,44,50,44,52,44,52,42,50,46,54,282,296,864,54,138,54,40,56,40,54,40,56,40,54,40,54,42,54,40,56,40,54,140,52,44,52,42,52,44,52,140,52,140,52,138,54,42,54,138,54,138,54,136,54,42,54,140,52,44,50,140,52,140,52,44,54,40,54,42,54,138,54,138,54,136,56,40,54,40,56,40,54,42,54,40,56,42,52,140,52,140,52,140,52,140,54,136,54,42,54,138,54,40,54,42,54,138,54,40,54,42,54,44,52,42,52,44,52,42,52,44,52,140,54,136,56,0

256dpi, 2016-07-08:
Hi @johnpica, just to be sure did you already change the following line in the library?

```
#define MQTT_BUFFER_SIZE 128
```

This is also explained in the Readme:

> The maximum size for packets being published and received is set by default to 128 bytes. To change that value, you need to download the library manually and change the value in the following file: https://github.com/256dpi/arduino-mqtt/blob/master/src/MQTTClient.h#L5.

johnpica, 2016-07-08:
Hi @256dpi 
Yes, I did. Everything is OK if I receive 1 message every 5 seconds. The problem is that when I receive 5-6 big messages in a second my app freeze.

johnpica, 2016-07-08:
@256dpi I think you can close the topic.. it was due to some bad placed delay().

Thank you.

256dpi, 2016-07-08:
Ok. Be ware that using `delay()` is never a good idea on a microcontroller. ;)

closed 2016-07-08 by 256dpi