256dpi/arduino-mqtt #49: Support for QoS1 and QoS2 publishing/subscribing

q2dg, 2016-08-21:
I know in Readme there is this pharagraph: "The first release of the library only supports QoS0 and the basic features to get going. In the next releases more of the features will be available. Please create an issue if you need a specific functionality." So, well...I'm creating the issue, then.

Thanks

PS: This alternative (https://github.com/adafruit/Adafruit_MQTT_Library) supports QoS1 publishing. This other (https://github.com/knolleary/pubsubclient) supports QoS1 subscribing

256dpi, 2016-08-22:
Thanks for raising this.

The underlying library (Paho) uses constants to activate QOS1 an QOS2 and as the Arduino IDE offers no library configuration via constants, enabling these features would result in more program space usage for all users.

Therefore, this feature is delayed until the Arduino IDE offers a suitable solution:
https://github.com/arduino/arduino-builder/issues/15

sandeepmistry, 2017-01-26:
@q2dg what boards would you like to use the QoS1 and QoS2 with?

256dpi, 2017-01-26:
Since the increase in programming space usage is minimal, I think we should enable support for all QoS levels in v2.

The usage table again for future reference:

```
QoS0: 25768 bytes (9%)
QoS1: 26240 bytes (10%)
QoS2: 26552 bytes (10%)
```

q2dg, 2017-01-27:
@sandeepmistry Hello! Well, I'm not a skilled technician but I think SAMD's boards (Zero and MKRxxx) -at least- could do the job, don't you think?

sandeepmistry, 2017-01-27:
@q2dg yes, even AVR boards like the Uno should be fine. See comments in https://github.com/256dpi/arduino-mqtt/issues/53#issuecomment-275456329 for more metrics.

alim-s-ajiev, 2017-04-26:
I am trying to use this library with ESP8266 and QoS1. I have not written any app code yet, just in selecting sutable library yet, but I suppose that some modification would be useful in src/lib/MQTTClient.h:

```
#if MQTTCLIENT_QOS1
#ifdef ESP8266
extern "C" void yield( void );
#endif
#endif

// only used in single-threaded mode where one command at a time is in process
template<class Network, class Timer, int a, int b>
int MQTT::Client<Network, Timer, a, b>::waitfor(int packet_type, Timer& timer)
{
    int rc = FAILURE;

    do
    {
        if (timer.expired())
            break; // we timed out
#ifdef ESP8266
        yield();
#endif
    }
    while ((rc = cycle(timer)) != packet_type);

    return rc;
}

```

Please, fix me if I am wrong!

256dpi, 2017-04-26:
@alim-s-ajiev The library has been tested and and successfully used with the ESP8266. No modification should be necessary to use with the current feature set.

alim-s-ajiev, 2017-04-26:
Yes, you are right! It turns out that yield() is called inside cycle method for ESP8266 WiFi.

Thank you very much!

256dpi, 2017-04-26:
This is beeing addressed in #59.

closed 2017-04-26 by 256dpi