256dpi/arduino-mqtt #52: Why should I use this instead PubSubClient?

felipemfp, 2016-12-04:
You know, I've been worked with the [PubSubClient](https://github.com/knolleary/pubsubclient) and now I find your library.

Is your library better? In what aspects?

256dpi, 2016-12-05:
The main differences are outlined in the readme:

> This library bundles the Embedded MQTT C/C++ Client library of the Eclipse Paho project and adds a thin wrapper to get an Arduino like API. Additionally there is an drop-in alternative for the Arduino Yùn that uses a python based client on the linux processor and a binary interface to lower program space usage on the Arduino side.

I wouldn't say one is better than the other, I guess it's just another approach.

felipemfp, 2016-12-05:
Thanks @256dpi!

closed 2016-12-05 by felipemfp

smitthhyy, 2017-10-27:
For others that may read this adding just two that where important for me:
- This client enables TLS over MQTT
- This client enables qos on publish