256dpi/arduino-mqtt #53: Feedback for V2

256dpi, 2017-01-26:
@gibix @sandeepmistry and all others.

I want to collect some ideas and thoughts about a possible V2 of this library. Do you have anything to add to what is already tagged with the [v2](https://github.com/256dpi/arduino-mqtt/issues?q=is%3Aopen+is%3Aissue+milestone%3Av2) milestone?

sandeepmistry, 2017-01-26:
@256dpi the list looks good so far!

I think it's partially covered by #39, however the Wire and new I2S Arduino libraries have a function to called `onReceive` to set a function handler for received data (https://www.arduino.cc/en/Reference/WireOnReceive). It would be nice to have this, instead of forcing sketched to define `void messageReceived(...)`.

256dpi, 2017-01-26:
@sandeepmistry That's a nice idea, will create a separate issue for that.

256dpi, 2017-01-26:
How do you feel about supporting QoS1 and QoS2? I still have the feeling it complicates things too much and might not be really used by the community.

sandeepmistry, 2017-01-26:
Do you have any benchmarks for the RAM and code space it adds?

256dpi, 2017-01-26:
Here are the numbers for the WiFi101 example on a MKR1000 board (without testing the code):

```
QoS0: 25768 bytes (9%)
QoS1: 26240 bytes (10%)
QoS2: 26552 bytes (10%)
```

Honestly, I tought the difference would be much bigger.

sandeepmistry, 2017-01-26:
Nice! Do you nice time to check the Uno as well? It should also print out the RAM used.

256dpi, 2017-01-26:
Here is the Arduino Uno with WiFi101:

```
QoS0: 25294 bytes (78%) - 1042 bytes (50%)
QoS1: 25980 bytes (80%) - 1042 bytes (50%) + 686 bytes
QoS2: 26540 bytes (82%) - 1092 bytes (53%) + 560 bytes
```

The differences seem to be slightly bigger.

sandeepmistry, 2017-01-26:
Ok, not bad!

Are 1042 and 1092 bytes for the RAM? If so, it sounds reasonable to go ahead with the feature :)

256dpi, 2017-04-26:
This is beeing addressed in #59.

closed 2017-04-26 by 256dpi