256dpi/arduino-mqtt #55: Allow dynamic setting of message received handler

256dpi, 2017-01-26:
As proposed by @sandeepmistry:

> [..], however the Wire and new I2S Arduino libraries have a function to called onReceive to set a function handler for received data (https://www.arduino.cc/en/Reference/WireOnReceive). It would be nice to have this, instead of forcing sketched to define void messageReceived(...).

tigoe, 2017-03-17:
I like this idea.  I would also suggest that the two extra parameters, (char * bytes, unsigned int length) be given defaults, since they're not used in the example sketches. Alternately maybe use an overloaded function signature?

If char * bytes is absolutely necessary, it'd be nice to make it a little more in line with the Arduino API style guide (https://www.arduino.cc/en/Reference/APIStyleGuide). I get around it by doing char bytes[]. 

I'm glad this is already on the milestone list. Thank you for making a very nice library!

256dpi, 2017-04-29:
This is beeing addressed in #59.

closed 2017-04-29 by 256dpi