256dpi/arduino-mqtt #60: How do I publish a retained message

nordicblue, 2017-02-23:
Sorry, I know that this is not really an issue with your code and more with my stupidity or lack  of knowledge. What I cannot figure out is how do I publish a retained message. 

I know how to publish a normal message, as:

client.publish(stateSpeedTopic + String(invId), (fanSpeedStatus));

Both stateSpeedTopic and fanSpeedStatus are of type string. 

How do I alter this to send a retained message instead?

Thank for so much for any help, I have been scratching my head for hours and not getting anywhere:(

P.S. Thank you so much for the code, it has been working excellent and is easier for me to folllow than the other alternative for esp8266 mqtt client available.  I just need to get the state of my fan to be retained in my MQTT server. Otherwise, my fan decides to turn on randomly...lol.

256dpi, 2017-02-23:
The current API does not provide a convenience function to publish a message with the retained flag. You have to use the more low-level function:

```c++
boolean publish(MQTTMessage * message);
```

The following is a snippet that explains the usage: 

```c++
String topic;
String payload;

MQTTMessage message;

message.topic = (char*)topic.c_str();
message.payload = (char*)topic.c_str();
message.length = topic.length();
message.retained = true;

client.publish(&message);
```

As this is a little complicated to figure out I will conside to simplify it in v2.

nordicblue, 2017-02-24:
Thanks for the quick reply. I'll try to play with that to get it working and I look forward to any future enhancements:)

closed 2017-02-24 by nordicblue