256dpi/arduino-mqtt #64: Make CONNACK packet's return code available to application

markrad, 2017-04-26:
Ideally one would be able to determine why a connection request failed. By simply returning false this makes it difficult to establish a recovery procedure. Consider the following scenario: I use the library WiFiManager to acquire a SSID, password and authorization information for a MQTT server. In the event that the WiFi connection fails, I throw the device back into AP mode giving the user an opportunity to retype the SSID and password. However, if the MQTT connection fails I have no idea if it was because the server was offline, not accessible from my location or the authorization information was incorrect. In the event of the latter, I want to throw it back into AP mode so the user can reenter their MQTT authorization whereas in the former two scenarios I simply want to retry the connection. The CONNACK packet clearly states the reason for connection failure but it is not exposed to the application. I'm aware that simply returning this value instead of the BOOL would be a breaking change for existing code. Perhaps you could make it available as a class instance getter such as?

```C
MQTTClient client;
client.begin(...);

if (!client.connect(id, name, password)
{
   if (client.getReturnCode() == MQTT_BADUSERNAME) // 0x04
   {
      // Get a new userid and password
    }
}
```

Thanks.

256dpi, 2017-04-26:
Indeed, that feature would be very helpful. I had a similar idea some time ago: #29.

This will be resolved with the development of v2.

markrad, 2017-04-26:
Thanks for responding.

Any ETA on v2?

256dpi, 2017-04-26:
This is beeing addressed in #59.

closed 2017-04-26 by 256dpi

markrad, 2017-04-27:
Glad to see you are addressing the buffer size issue too. Thanks. I look forward to it.