256dpi/arduino-mqtt #65: MQTT 3.1

roboticboyer, 2017-05-06:
I'm using a uMQTT Broker that runs on anESP8266
[https://github.com/martin-ger/esp_wifi_repeater/tree/uMQTT_broker](https://github.com/martin-ger/esp_wifi_repeater/tree/uMQTT_broker)

For the moment the Broker is compatible only with MQTT 3.1
How could I force arduino-mqtt library to use MQTT 3.1 and not 3.1.1?

Thanks

256dpi, 2017-05-06:
The library is currently enforcing MQTT 3.1.1 as it supports only that version.

However you can set the underyling Paho client to the 3.1 version in the connect method.

Add the following on this line: https://github.com/256dpi/arduino-mqtt/blob/master/src/MQTTClient.cpp#L70.

``` c++
this->options.MQTTVersion = 3;
```

Note: I haven't tested that. ;)

roboticboyer, 2017-05-13:
The patch is working
I suggest to add your note on the manual of the library
Thank You

roboticboyer, 2017-05-21:
I have written an article about my test with your library and the ESP8266 uMQTT Broker :
[http://wp.me/p4ZDu-3s](http://wp.me/p4ZDu-3s)

256dpi, 2017-05-22:
Cool, thanks! I'm closing this now.

closed 2017-05-22 by 256dpi