256dpi/arduino-mqtt #66: Multiple connections

martinvarbanov, 2017-05-09:
Hello,

Long story short, I'm trying to automize my program to make multiple connections to a Mosquitto brocker, without having to change the code every time I have to make a new connection. So far I've added 3 more connections from a single NodeMCU-ESP8266 with adding a **PubSubClient client(espClient)** and a **reconnect()** with just changing the name (client). My question is this: Is it possible to add some sort of counter (client **counter**), so when an event occurs, it to make a new connection?

256dpi, 2017-05-12:
When I understood your question properly you are asking if its possible to define multiple clients within the same Arduino sketch?

martinvarbanov, 2017-05-15:
Yes and is it possible to automize the process of defining multiple clients within the same Arduino sketch?

256dpi, 2017-05-15:
I will add this to the V2 milestone.

256dpi, 2017-06-28:
V2 (#59) will properly allow multiple clients.

closed 2017-06-28 by 256dpi