256dpi/arduino-mqtt #68: Support for QoS = 1

jasiek, 2017-06-26:
I think it would make a lot of sense to support QoS 1 in this client - from my understanding of the protocol, it would require:

* allocating storage space for tracking packets which are yet to be acknowledged (capacity should be configurable)
* retransmitting packets when no confirmation has been received

I think a good starting point would be retaining 10 last packet ids by default. Once a packet is confirmed, it is removed from the set.

Further refinements could add an exponential delay mechanism for retransmissions, etc.

Did I miss anything?

256dpi, 2017-06-28:
Thanks for your suggestions. Actually, QoS1 can be achived without any extra storage when the ack is immediately checked by waiting for it. (This is what the Paho and lwmqtt library do).

In anycase V2 (#59) will support QoS 1 and 2. Some earlier discussion can be found here #49.

I'll close this as it is a duplicate and already taken care of.

closed 2017-06-28 by 256dpi

jasiek, 2017-06-28:
Is anyone working on this? If not, I could look into it.

256dpi, 2017-06-28:
Yes, I'm just finishing the PR. I plan to release V2 this week. :)

In the meantime your are welcome to test the branch in your project. The API is fixed and the code stable. I'm just clearing up some docs.

jasiek, 2017-06-28:
Awesome. 👍 👍 👍 🥇

jasiek, 2017-06-29:
So which branch is it? I can only see -master in this repo.

256dpi, 2017-06-29:
I merge V2 yesterday. You can use master or wait until the library manager has been updated.