256dpi/arduino-mqtt #69: often disconnect?

weiyouren, 2017-07-13:
Reconnect every few minutes，logs:

Scan wifi ... 2 networks found
Connecting to 03405
........
WiFi connected
IP address: 192.168.0.110
network time: 6:27:13
MQTT connecting...connected
MQTT connecting...connected
MQTT connecting...connected
MQTT connecting...connected

256dpi, 2017-07-14:
What broker are you using?

weiyouren, 2017-07-14:
erlang mqtt broker emqttd https://github.com/emqtt/emqttd

madtoc, 2017-07-18:
is this the only client that you are connecting?

weiyouren, 2017-07-19:
yes

256dpi, 2017-07-20:
Do these clients have the same client-id? If so, they force any previous connection to be closed upon connection.

Teddyz, 2017-07-29:
Are the disconnects about 90 seconds apart? I found that most often, but not always, the client forgets to send PINGREQ. If you don't publish anything for 1.5  keep-alive period, the client is thrown out from the broker.

weiyouren, 2017-07-29:
@Teddyz i think you are right

Teddyz, 2017-07-29:
I let it idle for a few hours and in the log it can be seen that it mostly does not send PINGREQ.

[esp8266-pingreq-mosquito.txt](https://github.com/256dpi/arduino-mqtt/files/1185189/esp8266-pingreq-mosquito.txt)

On the positive side, it works flawlessly on Mega 2560 with a wired Ethernet shield. :)

256dpi, 2017-07-30:
Can you both maybe post the code that is causing the issues?

Is it possible that the `client.loop()` function is not called periodically in your code?

weiyouren, 2017-07-31:
`#include <MQTTClient.h>
#include <ArduinoJson.h>
#include <ESP8266httpUpdate.h>
#include "MyWifi.h"

#define HOST "iot.kechint.com"

const String PRODUCT_KEY = "test";
const String DEVICE_NAME = "ESP_" + String((long) ESP.getChipId());

const String TOPIC_REC    = PRODUCT_KEY + "/" + DEVICE_NAME + "/rec";
const String TOPIC_SEND   = PRODUCT_KEY + "/" + DEVICE_NAME + "/send";

const char* USERNAME = "user1";
const char* PASSWORD = "123456";

WiFiClientSecure wifi;
MQTTClient mqtt(256);
MyWifi myWifi;

const int KEY_PIN   = 2;
const int LED_PIN   = 12;
const int RELAY_PIN = 14;

void setup() {
  pinMode(LED_PIN, OUTPUT);
  pinMode(KEY_PIN, INPUT);
  pinMode(RELAY_PIN, OUTPUT);

  Serial.begin(115200);
  delay(100);

  LOGLN("\n");
  
  myWifi.init();

  mqtt.begin(HOST, 8883, wifi);
  mqtt.onMessage(mqttMessageReceived);
}

void mqttMessageReceived(String &topic, String &payload) {
  LOG("Message arrived [");
  LOG(topic);
  LOG("] ");
  LOGLN(payload);
}

void connect() {
  LOG("MQTT connecting...");
  while (!mqtt.connect(DEVICE_NAME.c_str(), USERNAME, PASSWORD)) {
    LOGLN(" try again in 5 seconds");
    delay(5000);
  }

  LOGLN("connected");

  mqtt.subscribe(TOPIC_REC.c_str(), 1);
  mqtt.subscribe(TOPIC_SEND.c_str(), 1);
}

void loop() {
  if (!mqtt.connected()) {
    connect();
  }

  mqtt.loop();
  delay(10);
}`

logs:
Scan wifi ... 1 networks found
Connecting to 91goz_1
........
WiFi connected
IP address: 192.168.1.106
MQTT connecting...connected
MQTT connecting...connected
MQTT connecting...connected

closed 2017-08-09 by weiyouren