256dpi/arduino-mqtt #71: problems with reconnection to broker

Teddyz, 2017-07-30:
If I turn off the Mosquitto broker and start it again, the Mega2560-client will not reconnect. The ESP8266 reconnects without a problem.

To be able to reconnect I had to add a 'disconnect' into the connection loop. Now my code looks like this:
```C
void mqConnect() {
  uint8_t loops = 0;
  while (!mq.connect(MQTT_CLIENTID, MQTT_USERNAME, MQTT_KEY)) {
    Serial.print('.');
    if (++loops >= 2)
      return;
    delay(200);
    mq.disconnect();
  }
  Serial.print("\nConnected to mqtt-broker\n");
...
```
On the ESP8266, if I run 'disconnect' without being connected, it reboots. 

I have all work-arounds I need now, but it would of course be nice to other beginners if this could be handled by the library.

256dpi, 2017-07-30:
Most likely there is some null pointer exception when calling `disconnect` while not beeing connected. This should be an easy fix in the library.

But the difference between the mega2560 and the esp8266 is weird, since the same code is executed on both platforms. Maybe we can check this again when the issue has been fixed. I'll work on this as soon as I'm back.

256dpi, 2017-08-18:
I fixed a bug that might have caused a crash when disconnecting without having any previous connection. Can you give the new `v2.1.0` version a try?

Teddyz, 2017-10-03:
```C
void mqConnect() {
  uint8_t loops = 0;
  toggle (24, HIGH);
  mq.setWill("node/" MQTT_CLIENTID "/status", "0", true, 0); // ... retained, QoS
  while (!mq.connect(MQTT_CLIENTID, MQTT_USERNAME, MQTT_KEY)) {
    Serial.print('.');
    if (++loops >= 2) {
      toggle (24, LOW);
      return;
    }
    delay(400);
    //mq.disconnect();
  }
  toggle (24, LOW);

  Serial.print(F("\nConnected to mqtt-broker\n"));

  mq.subscribe("node/#");
```
Tested on v2.1.4. On Mega2560 it still does not reconnect after broker has been offline for a while. (code above). If the broker is offline when Arduino boots up, Mega2560 has no problem connecting to the broker. If I uncomment  `mq.disconnect();` it works.

Other info:
-ESP8266 now sometimes 'hangs' after I disconnect from broker, use the `WiFiClientSecure` for another connection and then reconnects to the broker again. They hangs forever, so I physically have to disconnect power to them. The built in watchdog fails. This did not happen on v2.1.3 and earlier.
-One ESP8266 sent information very seldom, often forcing the library to `PINGRREQ`. This did not work when the client had been online for like 8 hours or more in previous versions. Instead the client was forced offline and then immediately reconnected. This is OK in v2.1.4.

giacomopastore, 2017-10-17:
Hello!

I'm working on Genuino MKR1000 (SAMD21 + ATWINC1500 WiFi module), I'm using connect and publish functions as described in the example.
If I stop/start MQTT broker (Mosquitto) the client is able to recconnects to the broker without problems but often, after some publish commands, the client looses the connection to the broker and it is not able to recover this state, on the server I've got this error:

Socket error on client <unknown>, disconnecting.

256dpi, 2017-10-17:
Hi all, I just pushed some commits that may fix the metioned problem (check the git history for what has been changed). It would be awesome if you could rerun your tests with master branch of the library before I release a new version

256dpi, 2017-10-18:
I just released v2.2.0 (might take some time to appear in the library manager) that fixes more related issues. Can you check again?

skedone, 2017-10-24:
@256dpi I have the same issue using the latest version (v2.2.0). 
Whenever a timeout disconnection happens, it doesn't reconnect.
```
1508819521: Client <IMEI> has exceeded timeout, disconnecting.
1508819521: Socket error on client <IMEI>, disconnecting.
```

Here is the code I use:

```c
void setup() {
  mqtt.begin(MQTT_SERVER, client);
  mqtt.setOptions(60, false, 2500);
  mqtt.onMessage(messageReceived);
  mqtt_subscribe();
}

void loop() {
  mqtt.loop();
  if(!mqtt.connected()) {
    mqtt_subscribe();
  }
  ....
}

bool mqtt_subscribe() {
  if (!mqtt.connected()) mqtt_connect();
  mqtt.subscribe("topic/p/" + IMEI, MQTT_QOS);
  mqtt_event_subscribe(); # sending events as soon as the device subscribes
  return true;
}

bool mqtt_connect() {
  if (!mqtt.connected()) {
    Serial.print("\nConnecting to MQTT server...");
    while (!mqtt.connect(IMEI.c_str(), MQTT_USER, MQTT_PASS)) {
      Serial.print(".");
      delay(500);
    }
    mqtt.setWill("topic/testament", IMEI.c_str());
    Serial.print(" [OK]");
  }

  return true;
}
```

It run for almost 38 hours without a glitch (thanks for the amazing library).
Any idea where to look into?

giacomopastore, 2017-10-24:
Now, with my hardware and v2.2.0 runs without problems. Thanks!

Teddyz, 2017-10-27:
The first issue, that the code refuse to reconnect when the broker is online again is solved in v2.2.0. Good work!
Second issue with ESP rebooting if a disconnect() is issued while having no connection on ESP8266 also passed the test. No restarts. 

Thanks!

closed 2017-10-27 by Teddyz