256dpi/arduino-mqtt #73: strange clientID

Urs-Eppenberger, 2017-08-16:
Hello Joël
I changed today from imroy's mqtt library to yours, since I have stuck connections where I suspect his library to be the culprit.
Unfortunately, I have an error with your library too, but another one.

Here is an extract from the mqtt broker log (mosquitto on RaspberryPi):
`1502914705:` New connection from 192.168.11.80 on port 8883.`
`1502914705: New client connected from 192.168.11.80 as u (c1, k60).`
It seems that the clientID is 'u'.
Since I use the same code for many devices, I fetch the hostname from a JSON formatted string stored in SPIFFS on the device. This hostname is definitely not 'u', since it is used at other places in the code and where it works. But since I'm not a programmer, but merely Frankensteining together foreign code, I might have made an error here. For your reference, I attach the full code below.
Many thanks for any insight.
Kind regards,
Urs.
[Sonoff_Generic_v1k.txt](https://github.com/256dpi/arduino-mqtt/files/1229439/Sonoff_Generic_v1k.txt)

Teddyz, 2017-08-17:
My best guess is that your code failed to read 'MQTT_User' from the json-file. 
I think the single u means that no username was provided. This is how it looks for me.
`1502949323: New client connected from 192.168.1.77 as mega2560 (c1, k60, u'mega2560').`

Good luck!

Urs-Eppenberger, 2017-08-17:
Thanks for the support. You have even looked at my code!
I analysed the mosquitto logfile closer. I have an entry from a working connection by the same code, but different MQTT library (imroy):
`1502516513: New client connected from 192.168.11.80 as sonoffth1 (c1, k15, u'mqtt').`
The MQTT clientID is placed before the bracket. The user (mqtt) is inside the bracket. The password is (luckily) not in the logfile. My conclusion is:
- the ClientID is either not sent or missing
- the username is missing too
Therefore the problem is even worse, but at least consistent. Maybe Joël has an idea.
Kind regards,
Urs.

256dpi, 2017-08-18:
Dear Urs,

I can ensure you that the arduino-mqtt library is correctly passing along the client-id, username and password parameters if they are correct C strings.

But in your code I can see that you are allocating the `StaticJsonBuffer<700>  jsonBuffer;` on the stack in `setup()`. Without knowing the ArduinoJSON library by heart I believe that the derived C strings from the JSON object are just pointers to parts of that buffer. In other words, the memory is reused when leaving `setup()` and entering `loop()`.  Now since 700 bytes are a big portion of memory the data might still be there, and thats why you probaly run into inconsistencies between the libraries as they affect the memory layout.

I hope that helps.

Best,
Joël

Urs-Eppenberger, 2017-08-18:
Hello Joël, thanks for your feedback and for having a look at my code.
 I started to suspect something like that yesterday. I'm not a programmer, especially not in C and really do not have a clue about the difference between pointers and why the heck something which is once in a variable just disappears sometimes later after a few hours of running. Really. This is stupid. But it just happened with my code.
Since I have no clue about programming, I said to myself, if I copy the variable content into another variable where there is no pointer strangeness, it might solve my problem. Here is what I did, most probably ugly as hell. I just copy the code section which handle the MQTT client ID:
In the top section:
`const char* Hostname;`
`String mqtt_id_s;`
`const char* mqtt_id_p;`
In the setup() section:
`StaticJsonBuffer<700>  jsonBuffer;`
`JsonObject& readc = jsonBuffer.parseObject(buf.get());`
`Hostname =  readc["Hostname"];`
`mqtt_id_s = String(Hostname);`
`mqtt_id_p = mqtt_id_s.c_str();`
And then where the MQTT connection to the broker is started:
`mqttClient.connect(mqtt_id_p, mqtt_user_p, mqtt_pass_p);`
If someone more knowledgeable than me on this topic (and there must be thousands out there) can help to make this into better code, please drop a note.
Kind regards,
Urs.

256dpi, 2017-08-18:
Converting the C string to a `String` object is a good approach as it will automatically allocate memory on the heap to store the data.

FYI: C has no support for string objects as other programming  languages do. A `const char *` just points to the address of the first byte of the string. By convention any string operation will stop if a null byte is found e.g. `\0`. Thus, if you copy a `const char *` you just copy an address and not its contents. That's why Arduino provides the `String` object.

closed 2017-08-18 by 256dpi

Urs-Eppenberger, 2017-08-18:
Thanks for your support, I really appreciate it.
With your last hint I'm now able to make the code a little prettier, now that it works and I even understand why.