256dpi/arduino-mqtt #74: Possibility to check fingerprint?

Teddyz, 2017-08-17:
Running SSL works fine on ESP8266. If you have time I would like the possibility to verify the servers SHA1 fingerprint. If I connect  to a server manually it looks like this:

`  const char* fingerprint = "A8:D6:B6:0F:8B:E2:F6:5B:EE:94:16:88:BA:9B:4D:BD:58:CC:C0:AC"; // https://www.grc.com/fingerprints.htm
...
  WiFiClientSecure  c;

  if (!c.connect(host, httpsPort)) {
    Serial.println("Secure connection to api.pushjet.io failed");
  }

  if (c.verify(fingerprint, host)) {
    Serial.println("certificate matches");
  } else {
    Serial.println("certificate doesn't match");
  }
...
`

Thanks!

256dpi, 2017-08-18:
The architecture of this lib is that the underlying network connection is passed to the library as a `Client`. This means that within the library we do not know if this is a secure connection or not and whether it implementes esp8266, esp32 or other platform specific functions. Therefore, we need to leave such things to the user. Since you have access to the client and network connection It should be possible to do it this way:

```cpp
Serial.print("\nconnecting...");
while (!client.connect("arduino", "try", "try")) {
  Serial.print(".");
  delay(1000);
}

if (net.verify(fingerprint, host)) {
  Serial.println("certificate matches");
} else {
  Serial.println("certificate doesn't match");
}

Serial.println("\nconnected!");
```

Can you give it a try?

Teddyz, 2017-08-20:
Works like a charm. 
Thanks for your help!

closed 2017-08-20 by Teddyz

esistgut, 2018-06-03:
Sorry to revive this old issue but I have a question about this: wouldn't this verify the fingerprint only *after* the connection is established and the clientid/user/password are transmitted over a possibly insececure channel?

256dpi, 2018-06-06:
@esistgut Good catch!

We could add an optional callback that is called after connecting the socket but before the connect packet is sent:

```c++
void onVerification(MQTTClient *client, Client net) {

}

client.setVerificationCallback(onVerification);
```

esistgut, 2018-06-06:
Or maybe just allow the developer to pass an already initialized WiFiClientSecure? I'm not sure about the internals of this library but please note that the fingerprint verification is no longer the only available security mechanism because it is now possible to verify the certificate chain https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WiFi/examples/HTTPSRequestCACert and there is a whole new SSL stack being implemented https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WiFi/examples (BearSSL stuff).

256dpi, 2018-06-13:
See #109 for a possible solution to the problem.