256dpi/arduino-mqtt #75: error in new library

Urs-Eppenberger, 2017-08-18:
Hello Joël
I use the Arduino IDE, got an announcement for your new version, updated it and recompiled my meanwhile running code.
I get an error while compiling since the library update:
`Build options changed, rebuilding all
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/client.c: In function 'lwmqtt_read_from_network':
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/client.c:76:72: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
         client->network_read(client, client->network, client->read_buf + offset + read, len - read, &partial_read, remaining_time);
                                                                        ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/client.c:76:81: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
         client->network_read(client, client->network, client->read_buf + offset + read, len - read, &partial_read, remaining_time);
                                                                                 ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/client.c: In function 'lwmqtt_write_to_network':
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/client.c:105:74: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
         client->network_write(client, client->network, client->write_buf + offset + written, len - written, &partial_write, remaining_time);
                                                                          ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/client.c:105:83: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
         client->network_write(client, client->network, client->write_buf + offset + written, len - written, &partial_write, remaining_time);
                                                                                   ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/client.c: In function 'lwmqtt_read_packet_in_buffer':
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/client.c:150:59: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
     err = lwmqtt_detect_remaining_length(client->read_buf + 1, len, &rem_len);
                                                           ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/helpers.c: In function 'lwmqtt_read_string':
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/helpers.c:22:35: warning: pointer of type 'void *' used in subtraction [-Wpointer-arith]
   unsigned long buf_len = buf_end - (*buf);
                                   ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/helpers.c:42:8: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
   *buf += str->len;
        ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/helpers.c: In function 'lwmqtt_write_string':
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/helpers.c:61:8: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
   *buf += string.len;
        ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/helpers.c: In function 'lwmqtt_read_num':
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/helpers.c:72:8: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
   *buf += 2;
        ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/helpers.c: In function 'lwmqtt_write_num':
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/helpers.c:86:8: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
   *buf += 2;
        ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/helpers.c: In function 'lwmqtt_read_byte':
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/helpers.c:94:8: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
   *buf += 1;
        ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/helpers.c: In function 'lwmqtt_write_byte':
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/helpers.c:107:8: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
   *buf += 1;
        ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/helpers.c: In function 'lwmqtt_read_varnum':
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/helpers.c:166:8: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
   *buf += len;
        ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/helpers.c: In function 'lwmqtt_write_varnum':
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/helpers.c:196:8: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
   *buf += len;
        ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/packet.c: In function 'lwmqtt_encode_connect':
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/packet.c:166:9: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
     ptr += will->message.payload_len;
         ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/packet.c:180:20: warning: pointer of type 'void *' used in subtraction [-Wpointer-arith]
   *len = (int)(ptr - buf);
                    ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/packet.c: In function 'lwmqtt_encode_zero':
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/packet.c:236:20: warning: pointer of type 'void *' used in subtraction [-Wpointer-arith]
   *len = (int)(ptr - buf);
                    ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/packet.c: In function 'lwmqtt_encode_ack':
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/packet.c:294:20: warning: pointer of type 'void *' used in subtraction [-Wpointer-arith]
   *len = (int)(ptr - buf);
                    ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/packet.c: In function 'lwmqtt_decode_publish':
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/packet.c:339:23: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
   void *end_ptr = ptr + rem_len;
                       ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/packet.c:352:17: warning: pointer of type 'void *' used in subtraction [-Wpointer-arith]
     if (end_ptr - ptr < 2) {
                 ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/packet.c:362:32: warning: pointer of type 'void *' used in subtraction [-Wpointer-arith]
   *payload_len = (int)(end_ptr - ptr);
                                ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/packet.c: In function 'lwmqtt_encode_publish':
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/packet.c:411:7: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
   ptr += payload_len;
       ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/packet.c:414:20: warning: pointer of type 'void *' used in subtraction [-Wpointer-arith]
   *len = (int)(ptr - buf);
                    ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/packet.c: In function 'lwmqtt_encode_subscribe':
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/packet.c:460:20: warning: pointer of type 'void *' used in subtraction [-Wpointer-arith]
   *len = (int)(ptr - buf);
                    ^
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/packet.c: In function 'lwmqtt_encode_unsubscribe':
/Users/eppen/Work_Private/Heimautomatisation/Arduino_Stuff/libraries/MQTT/src/lwmqtt/packet.c:552:20: warning: pointer of type 'void *' used in subtraction [-Wpointer-arith]
   *len = (int)(ptr - buf);
                    ^`

256dpi, 2017-08-18:
For which board are you compiling? The Arduino IDE does not show these errors in my builds...

256dpi, 2017-08-19:
I just published `v2.1.1` that should fix the issue. Can you confirm?

Teddyz, 2017-08-20:
v2.1.1 does not compile for ESP8266. v2.1.0 works. I have not tested to compile for Mega2560 yet.

Linking everything together...
---removed loong line---
D:\Temp\arduino_build_18058\libraries\MQTT\lwmqtt\string.c.o: In function "lwmqtt_strcmp":

C:\Users\Admin\Documents\Arduino\libraries\MQTT\src\lwmqtt/string.c:12: multiple definition of "lwmqtt_strcmp"

D:\Temp\arduino_build_18058\libraries\MQTT\lwmqtt\helpers.c.o:C:\Users\Admin\Documents\Arduino\libraries\MQTT\src\lwmqtt/helpers.c:7: first defined here

D:\Temp\arduino_build_18058\sketch\mqtt_vrum_v003.ino.cpp.o:(.text._Z8setup_mqv+0x20): undefined reference to `lwmqtt_arduino_network_read(lwmqtt_client_t*, void*, unsigned char*, unsigned int, unsigned int*, int)'

D:\Temp\arduino_build_18058\sketch\mqtt_vrum_v003.ino.cpp.o:(.text._Z8setup_mqv+0x24): undefined reference to `lwmqtt_arduino_network_write(lwmqtt_client_t*, void*, unsigned char*, unsigned int, unsigned int*, int)'

collect2.exe: error: ld returned 1 exit status

Urs-Eppenberger, 2017-08-20:
I compile for the "Generic ESP8266 Module" for my zoo of ESP8266 based devices.
I work with the Arduino IDE 1.8.2. on Mac OS X Sierra 10.12.6.
2.0.0 works
2.1.0 shows the above mentioned errors
2.1.1 works
As far as I am concerned, everything is fine. Many thanks and kind regards, Urs.

Teddyz, 2017-08-20:
Now I have tested Mega2560. It is a quite large sketch, only leaving around 1700 bytes of 8 kB available RAM.

**2.0.0** works.
**2.1.0** compiles, but lose connection after a second and does not reconnect. I get the feeling it has to do with memory management as I get slightly different behaviour if I omit some subscriptions. Also, **2.0.0** does not trigger PINGREQ as seen below.

1503212763: New client connected from 192.168.1.77 as mega2560 (c1, k60, u'Mega2560').
1503212763: Sending CONNACK to mega2560 (0, 0)
1503212763: Received SUBSCRIBE from mega2560
1503212763:     node/# (QoS 0)
1503212763: mega2560 0 node/#
1503212763: Sending SUBACK to mega2560
1503212763: Sending PUBLISH to mega2560 (d0, q0, r1, m0, 'node/kök/status', ... (1 bytes))
1503212763: Sending PUBLISH to mega2560 (d0, q0, r1, m0, 'node/kök/compiled', ... (20 bytes))
---some 20 more lines here---
1503212763: Received PUBLISH from mega2560 (d0, q0, r1, m0, 'node/mega2560/uptime', ... (1 bytes))
1503212763: Sending PUBLISH to mega2560 (d0, q0, r0, m0, 'node/mega2560/uptime', ... (1 bytes))
1503212763: Received PINGREQ from mega2560
1503212763: Sending PINGRESP to mega2560
1503212763: Received PINGREQ from mega2560
1503212763: Sending PINGRESP to mega2560
1503212764: Socket error on client mega2560, disconnecting.

**2.1.1** same as **2.1.0**

256dpi, 2017-08-20:
@Urs-Eppenberger: Good to hear that it is working for you again.

@Teddyz: Can you maybe flash the examples to your mega2560 and esp8266 device to check if the library itself is working? As you said yourself the later issue might be caused by the largeness of the sketch.

256dpi, 2017-10-18:
@Teddyz I just released v2.2.0 (might take some time to appear in the library manager). Can you check again?

Teddyz, 2017-10-27:
The issues I have might be related to long char-arrays. `MQTTClient mq(256);` might be what I needed. I have implemented a log file journal so I can see what happened when. Upcoming reports will be more specific. :)
Question: Is there any protection in the library against too long char-arrays or Strings?

256dpi, 2017-10-28:
Yes, if you receive to long messages or try to send to long messages the underlying library will return `LWMQTT_BUFFER_TOO_SHORT` and cause the client to disconnect. It then tries to reconnect again.

Teddyz, 2017-11-14:
I've been running v2.2.0 for a few days. I made a log file system on the ESP8266. In the loop it checks for lwmqtt errors and return codes. The ESP is close to a Wifi repeater that is wirelessly connected to a router far away. A few messages are sent every minute (rssi, uptime,...) The main purpose is to send alarm messages from the attached PIR-detector. QoS is always 0. 
[soprum_log.txt](https://github.com/256dpi/arduino-mqtt/files/1472648/soprum_log.txt)

I will now update to v2.2.1 and see if the log looks similar after some time.

It would be great if you could explain how we could make use of the boolean return-codes that many functions have.
Since my memory is terrible it would be good if there were a way to read the library version and put it in the log at every start. :)

closed 2018-02-08 by 256dpi