256dpi/arduino-mqtt #76: Send/Recived

luisrival, 2017-08-19:
Hi, thanks for the correction before, but it still not working, can you help me again saying what is wrong with my code? I want to send the status of one GPIO and recived one message and turn on an other GPIO with this comand but I actually send the status but I never recived the message. This is the code:

#include <ESP8266WiFi.h>
#include <MQTTClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <ESP8266HTTPUpdateServer.h>

/* ---------- DO NOT EDIT ANYTHING ABOVE THIS LINE ---------- */

//Only edit the settings in this section

/* WIFI Settings */
// Name of wifi network
const char* ssid = "Casa 2.4";

// Password to wifi network
const char* password = "%Cvap1992"; 

/* Web Updater Settings */
// Host Name of Device
const char* host = "MK-DoorSensor1";

// Path to access firmware update page (Not Neccessary to change)
const char* update_path = "/firmware";

// Username to access the web update page
const char* update_username = "luisrival@gmail.com";

// Password to access the web update page
const char* update_password = "$Olrvp3527";

/* MQTT Settings */
// Topic which listens for commands
char* outTopic = "MK-SmartHouse/security/MK-DoorSensor1"; 

//MQTT Server IP Address
const char* server = "192.168.15.208";

//Unique device ID 
const char* mqttDeviceID = "MK-SmartHouseDevice1"; 

/* ---------- DO NOT EDIT ANYTHING BELOW THIS LINE ---------- */

//the time when the sensor outputs a low impulse
long unsigned int lowIn;         

//the amount of milliseconds the sensor has to be low 
//before we assume all detection has stopped
long unsigned int pause = 100;  

//sensor variables
boolean lockLow = true;
boolean takeLowTime;  

//the digital pin connected to the door sensor's output
int sensorPin = 12;  

//webserver 
ESP8266WebServer httpServer(80);
ESP8266HTTPUpdateServer httpUpdater;

//MQTT 
WiFiClient net;
MQTTClient client;

//Time Variable
unsigned long lastMillis = 0;
int channel1 = 13;
int channel3 = 16;
//Connect to WiFI and MQTT
void connect();

//Setup pins, wifi, webserver and MQTT
void setup() 
{
  pinMode(sensorPin, INPUT);
  digitalWrite(sensorPin, LOW);
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  
  WiFi.begin(ssid, password);
  client.begin(server, net);
  client.subscribe(outTopic);
  client.onMessage(messageReceived);

  connect();

  MDNS.begin(host);

  httpUpdater.setup(&httpServer, update_path, update_username, update_password);
  httpServer.begin();

  MDNS.addService("http", "tcp", 80);
}

//Connect to wifi and MQTT
void connect() 
{
  while (WiFi.status() != WL_CONNECTED) 
  {
    delay(1000);
  }

  while (!client.connect(mqttDeviceID)) 
  {
    delay(1000);
  }
}

void loop() 
{
  // MQTT Loop
  client.loop();
  delay(10);

  // Make sure device is connected
  if(!client.connected()) 
  {
    connect();
  }

  httpServer.handleClient();

  //Sensor Detection
  if(digitalRead(sensorPin) == HIGH)
  {
    if(lockLow)
    {  
      //makes sure we wait for a transition to LOW before any further output is made:
      lockLow = false;            
      client.publish(outTopic, "OPEN");  
      delay(50);
    }         
    takeLowTime = true;
  }

  if(digitalRead(sensorPin) == LOW)
  {       
    if(takeLowTime)
    {
      lowIn = millis();          //save the time of the transition from high to LOW
      takeLowTime = false;       //make sure this is only done at the start of a LOW phase
    }
    //if the sensor is low for more than the given pause, 
    //we assume that no more detection is going to happen
    if(!lockLow && millis() - lowIn > pause)
    {  
      //makes sure this block of code is only executed again after 
      //a new detection sequence has been detected
      lockLow = true;                        
      client.publish(outTopic, "CLOSED");
      delay(50);
    }
  }
   
}
void messageReceived(String &topic, String &payload) 
{
 String msgString = payload;
   Serial.println("incoming: " + topic + " - " + payload);
  if (msgString == "Z1ON")
  {
    digitalWrite(channel1,1);
    delay(250);
  }
  else if (msgString == "Z1OFF")
  {
    digitalWrite(channel1,0);
    delay(250);
  }
  else if (msgString == "Z2ON")
  {
    digitalWrite(LED_BUILTIN,HIGH);
    delay(250);
  }
  else if (msgString == "Z2OFF")
  {
    digitalWrite(LED_BUILTIN,LOW);
    delay(250);
  }
  else if (msgString == "Z3ON")
  {
    digitalWrite(channel3,1);
    delay(250);
  }
  else if (msgString == "Z3OFF")
  {
    digitalWrite(channel3,0);
    delay(250);
  } //This sensor does not recieve anything from MQTT Server so this is blank
}

256dpi, 2017-08-19:
You need to put `client.subscribe(outTopic);` after `client.connect()`. A subscription can only be issued if the client is connected to the broker.

luisrival, 2017-08-19:
I understand that a subcription can only be  issued if the client is connected. but I don´t identify were is de client.connect() Because if I do like this, is the same result

#include <ESP8266WiFi.h>
#include <MQTTClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <ESP8266HTTPUpdateServer.h>

/* ---------- DO NOT EDIT ANYTHING ABOVE THIS LINE ---------- */

//Only edit the settings in this section

/* WIFI Settings */
// Name of wifi network
const char* ssid = "Casa 2.4";

// Password to wifi network
const char* password = "%Cvap1992"; 

/* Web Updater Settings */
// Host Name of Device
const char* host = "MK-DoorSensor1";

// Path to access firmware update page (Not Neccessary to change)
const char* update_path = "/firmware";

// Username to access the web update page
const char* update_username = "luisrival@gmail.com";

// Password to access the web update page
const char* update_password = "$Olrvp3527";

/* MQTT Settings */
// Topic which listens for commands
char* outTopic = "MK-SmartHouse/security/MK-DoorSensor1"; 

//MQTT Server IP Address
const char* server = "192.168.15.208";

//Unique device ID 
const char* mqttDeviceID = "MK-SmartHouseDevice1"; 

/* ---------- DO NOT EDIT ANYTHING BELOW THIS LINE ---------- */

//the time when the sensor outputs a low impulse
long unsigned int lowIn;         

//the amount of milliseconds the sensor has to be low 
//before we assume all detection has stopped
long unsigned int pause = 100;  

//sensor variables
boolean lockLow = true;
boolean takeLowTime;  

//the digital pin connected to the door sensor's output
int sensorPin = 12;  

//webserver 
ESP8266WebServer httpServer(80);
ESP8266HTTPUpdateServer httpUpdater;

//MQTT 
WiFiClient net;
MQTTClient client;

//Time Variable
unsigned long lastMillis = 0;
int channel1 = 13;
int channel3 = 16;
//Connect to WiFI and MQTT
void connect();

//Setup pins, wifi, webserver and MQTT
void setup() 
{
  pinMode(sensorPin, INPUT);
  digitalWrite(sensorPin, LOW);
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  
  WiFi.begin(ssid, password);
  client.begin(server, net);
  client.onMessage(messageReceived);

  connect();

  MDNS.begin(host);

  httpUpdater.setup(&httpServer, update_path, update_username, update_password);
  httpServer.begin();

  MDNS.addService("http", "tcp", 80);
}

//Connect to wifi and MQTT
void connect() 
{
  while (WiFi.status() != WL_CONNECTED) 
  {
    delay(1000);
  }

  while (!client.connect(mqttDeviceID)) 
  {
    client.subscribe(outTopic);
    delay(1000);
  }
}

void loop() 
{
  // MQTT Loop
  client.loop();
  delay(10);

  // Make sure device is connected
  if(!client.connected()) 
  {
    connect();
  }

  httpServer.handleClient();

  //Sensor Detection
  if(digitalRead(sensorPin) == HIGH)
  {
    if(lockLow)
    {  
      //makes sure we wait for a transition to LOW before any further output is made:
      lockLow = false;            
      client.publish(outTopic, "OPEN");  
      delay(50);
    }         
    takeLowTime = true;
  }

  if(digitalRead(sensorPin) == LOW)
  {       
    if(takeLowTime)
    {
      lowIn = millis();          //save the time of the transition from high to LOW
      takeLowTime = false;       //make sure this is only done at the start of a LOW phase
    }
    //if the sensor is low for more than the given pause, 
    //we assume that no more detection is going to happen
    if(!lockLow && millis() - lowIn > pause)
    {  
      //makes sure this block of code is only executed again after 
      //a new detection sequence has been detected
      lockLow = true;                        
      client.publish(outTopic, "CLOSED");
      delay(50);
    }
  }
   
}
void messageReceived(String &topic, String &payload) 
{
 String msgString = payload;
   Serial.println("incoming: " + topic + " - " + payload);
  if (msgString == "Z1ON")
  {
    digitalWrite(channel1,1);
    delay(250);
  }
  else if (msgString == "Z1OFF")
  {
    digitalWrite(channel1,0);
    delay(250);
  }
  else if (msgString == "Z2ON")
  {
    digitalWrite(LED_BUILTIN,HIGH);
    delay(250);
  }
  else if (msgString == "Z2OFF")
  {
    digitalWrite(LED_BUILTIN,LOW);
    delay(250);
  }
  else if (msgString == "Z3ON")
  {
    digitalWrite(channel3,1);
    delay(250);
  }
  else if (msgString == "Z3OFF")
  {
    digitalWrite(channel3,0);
    delay(250);
  } //This sensor does not recieve anything from MQTT Server so this is blank
}

256dpi, 2017-08-20:
You put it in the wrong place. That should work:

```c++
//Connect to wifi and MQTT
void connect() {
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
  }

  while (!client.connect(mqttDeviceID)) {
    delay(1000);
  }

  // the client is now connected
  client.subscribe(outTopic);
}
```

luisrival, 2017-08-20:
Ok thanks i will try

luisrival, 2017-08-21:
sorry but istill didn´t work, I send perfect de message of OPEN or CLOSED but I can´t recived.

//the time when the sensor outputs a low impulse
long unsigned int lowIn;         

`//the time when the sensor outputs a low impulse
long unsigned int lowIn;         

//the amount of milliseconds the sensor has to be low 
//before we assume all detection has stopped
long unsigned int pause = 100;  

//sensor variables
boolean lockLow = true;
boolean takeLowTime;  

//the digital pin connected to the door sensor's output
int sensorPin = 12;  

//webserver 
ESP8266WebServer httpServer(80);
ESP8266HTTPUpdateServer httpUpdater;

//MQTT 
WiFiClient net;
MQTTClient client;

//Time Variable
unsigned long lastMillis = 0;
int channel3 = 16;
//Connect to WiFI and MQTT
void connect();

//Setup pins, wifi, webserver and MQTT
void setup() 
{
  pinMode(sensorPin, INPUT);
  digitalWrite(sensorPin, LOW);
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  
  WiFi.begin(ssid, password);
  client.begin(server, net);
  client.onMessage(messageReceived);

  connect();

  MDNS.begin(host);

  httpUpdater.setup(&httpServer, update_path, update_username, update_password);
  httpServer.begin();

  MDNS.addService("http", "tcp", 80);
}

//Connect to wifi and MQTT
void connect() 
{
  while (WiFi.status() != WL_CONNECTED) 
  {
    delay(1000);
  }

  while (!client.connect(mqttDeviceID)) 
  {
    delay(1000);
  }
  client.subscribe(outTopic);
}

void loop() 
{
  // MQTT Loop
  client.loop();
  delay(10);

  // Make sure device is connected
  if(!client.connected()) 
  {
    connect();
  }

  httpServer.handleClient();

  //Sensor Detection
  if(digitalRead(sensorPin) == HIGH)
  {
    if(lockLow)
    {  
      //makes sure we wait for a transition to LOW before any further output is made:
      lockLow = false;            
      client.publish(outTopic, "OPEN");  
      delay(50);
    }         
    takeLowTime = true;
  }

  if(digitalRead(sensorPin) == LOW)
  {       
    if(takeLowTime)
    {
      lowIn = millis();          //save the time of the transition from high to LOW
      takeLowTime = false;       //make sure this is only done at the start of a LOW phase
    }
    //if the sensor is low for more than the given pause, 
    //we assume that no more detection is going to happen
    if(!lockLow && millis() - lowIn > pause)
    {  
      //makes sure this block of code is only executed again after 
      //a new detection sequence has been detected
      lockLow = true;                        
      client.publish(outTopic, "CLOSED");
      delay(50);
    }
  }
}
void messageReceived(String &topic, String &payload) 
{
 String msgString = payload;
   Serial.println("incoming: " + topic + " - " + payload);
  if (msgString == "Z1ON")
  {
    digitalWrite(2, 1);
    delay(250);
  }
  else if (msgString == "Z1OFF")
  {
    digitalWrite(2, 0);
    delay(250);
  }
  else if (msgString == "Z2ON")
  {
    digitalWrite(LED_BUILTIN,1);
    delay(250);
  }
  else if (msgString == "Z2OFF")
  {
    digitalWrite(LED_BUILTIN,0);
    delay(250);
  }
  else if (msgString == "Z3ON")
  {
    digitalWrite(channel3,1);
    delay(250);
  }
  else if (msgString == "Z3OFF")
  {
    digitalWrite(channel3,0);
    delay(250);
  } //This sensor does not recieve anything from MQTT Server so this is blank
}`

luisrival, 2017-08-21:
perfect!!! thanks!!!!

256dpi, 2017-08-21:
I guess it works now. I'll close this. :)

closed 2017-08-21 by 256dpi