256dpi/arduino-mqtt #77: 2.1.1 - no payload passed for onMessage/onMessageAdvanced

crathje, 2017-08-22:
Hi,

I just tried the library and had some issues with no payload (length = 0 and empty string) for the `onMessage` and `onMessageAdvanced` callbacks with 2.1.1.
Switching back to 2.1.0 solves the problem for me (ESP8266).

```cpp
void mqttMessageReceived(String &topic, String &payload) {
  Serial.println("incoming: " + topic + " - " + payload + " (" + String(payload.length()) + ")");
}
```
always yields
> incoming: /hello - (0)

with 2.1.0 it is working fine:
> incoming: /hello - shiftr testing (14)

Anyone else experienced such problems?

256dpi, 2017-08-22:
Thanks for reporting this. I identified the issue and will release a bugfix within the next hours.

256dpi, 2017-08-22:
I just released v2.1.3. It might take some time until it comes up in the library manager for download.

crathje, 2017-08-23:
I just tried v2.1.3 - works again.

Thanks for the fast support/bugfix.

closed 2017-08-23 by crathje

phretor, 2019-03-01:
I've noticed a slightly similar behavior on the latest version (not master, just pulling it via PlatformIO). If I send, say 4 bytes of payload from a client, and use `void mqttMessageReceived(String &topic, String &payload)`, `payload.length()` is only 1 byte, whereas using the advanced callback, the `payload_length` is filled with the correct size (4 bytes).

256dpi, 2019-03-01:
Is it possible that your payload contains a null character `0x0`?

phretor, 2019-03-01:
@256dpi hard to say, I'm (ab)using MQTT as a transport for nanopb-serialized data, so, could happen.