256dpi/arduino-mqtt #79: ESP reset with blank message from subscribed topic

mithunrockey, 2017-08-30:
Hello Joel,

I have been playing with your lib for past couple of days. All works fine till now I got a block on the road now.

Happens only when I try to post a blank message to the subscribed topic on ESP.

Here is the output.

Exception (29):
epc1=0x40202e4f epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff0250 end: 3fff0540 offset: 01a0

>>>stack>>>
3fff03f0:  3fff0430 3fff130e 3ffef22c 4020a644  
3fff0400:  6f546e69 00636970 4020a384 402090c1  
3fff0410:  3fff0480 3fff048c 3fff0470 40201a21  
3fff0420:  00000009 00000030 3fff1317 40202b46  
3fff0430:  3fff048e 00000000 00000000 402044cc  
3fff0440:  3fff130e 3fff130d 3ffef230 0000000b  
3fff0450:  3fff04c0 00000000 3ffef238 4020110a  
3fff0460:  00000000 0047f700 00000000 00000000  
3fff0470:  00000000 0047f700 00000000 00000000  
3fff0480:  40100007 3fff1310 0047f748 00000000  
3fff0490:  0000000b 00000000 00000000 4020936e  
3fff04a0:  3ffeeaf0 3ffef394 3ffef234 3ffef514  
3fff04b0:  3ffef438 0000000b 3ffef238 4020121c  
3fff04c0:  00000003 0000000a 3ffef50c 4020a5aa  
3fff04d0:  3ffef438 3ffef238 3ffef1dc 4020320d  
3fff04e0:  00000000 00000000 3ffe86e2 00000006  
3fff04f0:  000003e8 feefeffe 3ffef234 4020906c  
3fff0500:  00000000 00000000 3ffe86e2 00000006  
3fff0510:  00000000 00000000 00000001 4020a3a5  
3fff0520:  3fffdad0 00000000 3ffef50c 4020a3d0  
3fff0530:  feefeffe feefeffe 3ffef520 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vc2a0c112
~ld

The problem fires only if the callback is defined 
client.onMessage(messageReceived);

Can you check this issue and suggest me a fix for this. Thanks...

mithunrockey, 2017-09-01:
Just realized you are out for vacation. Sorry to bug you, please can you have a look at this when you are back.

256dpi, 2017-10-18:
I just released v2.2.0 (might take some time to appear in the library manager) that should fix this issues. Can you test again?

256dpi, 2017-11-15:
Closing, as I believe the issue has been resolved.

closed 2017-11-15 by 256dpi