256dpi/arduino-mqtt #80: Not working on Arduino Industrial 101

naxmon, 2017-08-31:
Hi everybody, 

has anyone tried the library on an Industrial 101? I don't really know if the library should work on this platform, but being the Industrial 101 a smaller YUN it should work right?
I tried both YUN examples and unfortunately they don't work properly. With the plain version, it connects to the broker and instantly disconnects. With the SSL example, the board never gets to connect to the broker. 
I updated the Linino to the latest version (02 06 2016) but it doesn't make things better.
Has someone an idea?

Any help would be appreciated.
Regards,
Nicolas

256dpi, 2017-10-18:
I recently discovered a bug with the Yun that has been fixed in the latest published version. Do you mind check this again (I don't have a Industrial 101 at hand)?

256dpi, 2017-12-06:
I'm closing this due to its age. Please reopen if the problem still exists.

closed 2017-12-06 by 256dpi