256dpi/arduino-mqtt #85: onMessage Callback "echoes"

mathieu-mp, 2017-11-15:
First, I'm not sure if this is a bug from the library of a mistake inside my code.

In the code below, I subscribe to a topic and correctly receive all the published messages from the subscribed topic...
...but there are echoes!

Example output of received messages. The publisher publishes value A __once__, __then__ value B __once__, __then__ value C __once__ :
```
jour/chauffage/sol/set/consigne: A
jour/chauffage/sol/set/consigne: A
jour/chauffage/sol/set/consigne: B
jour/chauffage/sol/set/consigne: A
jour/chauffage/sol/set/consigne: B
jour/chauffage/sol/set/consigne: A
jour/chauffage/sol/set/consigne: A
jour/chauffage/sol/set/consigne: B
jour/chauffage/sol/set/consigne: C
jour/chauffage/sol/set/consigne: A
jour/chauffage/sol/set/consigne: ...
```

Trigger of unwanted callbacks looks random to me, not strictly associated to the publishing of next value. Of course, the more received messages, the more callback echoes.

I checked with a debian client and I am sure the issue does not come from the publisher.
The message is retained and qos 2.

Here is my code, the onMessage callback is made from within the network class.

```c++
#include "Arduino.h"
#include "Network.h"

//WiFi
#include <ESP8266WiFi.h>          //ESP8266 Core WiFi Library (you most likely already have this in your sketch)

#include <DNSServer.h>            //Local DNS Server used for redirecting all requests to the configuration portal
#include <ESP8266WebServer.h>     //Local WebServer used to serve the configuration portal
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager WiFi Configuration Magic

#include <MQTTClient.h>           //Librairie MQTT

Network::Network() : _mqtt(), _wifiClient() {
  //TODO: Eventually read settings from SPIFFS
}

void Network::connectWifi() {
  //In case we need to erase previous config
  //WiFi.disconnect(true);

  //Nothing to do, the ESP reconnects automatically with previous credentials
  //WiFi.begin();
  //WiFi.begin(ssid,password);

  //MQTT
  beginMqtt();
}

void Network::disconnectWifi() {
  // Disconnects and erases the credentials
  WiFi.disconnect();
}

bool Network::isWifiConnected() {
  return _wifiConnected;
}

byte Network::getWifiConnectionStatus() {
  //WiFi.status():
  //0 : WL_IDLE_STATUS when Wi-Fi is in process of changing between statuses
  //1 : WL_NO_SSID_AVAILin case configured SSID cannot be reached
  //3 : WL_CONNECTED after successful connection is established
  //4 : WL_CONNECT_FAILED if password is incorrect
  //6 : WL_DISCONNECTED if module is not configured in station mode
  return WiFi.status();
}

void Network::beginOTA() {
  /*
  ...
  */
}

IPAddress Network::getIP() {
  return WiFi.localIP();
}

void Network::startOTA() {
  _otaEnabled = true;
}

void Network::stopOTA() {
  _otaEnabled = false;
}

void Network::startAP() {
  /*
  ...
  */
}

void Network::beginMqtt() {
  _mqtt.setWill("jour/chauffage/sol/statut","offline",true,2);
  _mqtt.begin("jeedom.vt.lan", _wifiClient);
  _mqtt.onMessage([](String &topic, String &payload) {
    Serial.println(topic + ": " + payload);
  });
}

void Network::connectMqtt() {
  _mqtt.connect("Thermostat");
}

bool Network::isMqttConnected() {
  return _mqtt.connected();
}

void Network::publishMqttTopic(String topic, String payload, bool retained, int qos) {
  _mqtt.publish(topic, payload, retained, qos);
}

void Network::subscribeMqttTopic(String topic, int qos) {
  _mqtt.subscribe(topic,qos);
}

void Network::loop() {
  yield();

  if(WiFi.status() != WL_CONNECTED) {
    _wifiConnected = false;
  } else {
    _wifiConnected = true;
    //OTA
    if (_otaEnabled) {
      ArduinoOTA.handle();
      delay(10);
    }
    //MQTT
    
    if (!isMqttConnected()) {
      connectMqtt();
      
      publishMqttTopic("jour/chauffage/sol/statut","online", true, 2);
      subscribeMqttTopic("jour/chauffage/sol/set/consigne", 2);
    }
    _mqtt.loop();
    delay(10);
    
  }
}
```

And here is the main code lauching the network stuff:
```c++
#include <Wire.h>
#include <Arduino.h>

//Classes locales
#include "Network.h"

Network network = Network();

void setup() {
  Serial.begin(115200);
  Serial.println("Initializing...");
  
  network.connectWifi();
  network.beginOTA();
  network.startOTA();
}

void loop() {
  network.loop();
}
```

256dpi, 2017-11-15:
There was a bug in the underlying library about wrong QoS flows. Your issue might be cuased by that. Can you test with the latest version I released yesterday?

mathieu-mp, 2017-11-15:
@256dpi, this solved the issue! Never seen such a fast patching, even before the issue was opened ;-)
**Thank your for that library, nice work, and great doc!**

Well, I know this is an issue tracker and not a forum, but a this point I am unable to assign the payload to a member variable from within my callback. Code below:
```c++
void Network::beginMqtt() {
  _mqtt.setWill(MQTT_TOPIC_WILL,"offline",true,2);
  _mqtt.begin("jeedom.vt.lan", _wifiClient);
  _mqtt.onMessage([](String &topic, String &payload) {
    Serial.println(topic + ": " + payload);
    if (topic.equals(MQTT_TOPIC_SET_CONSIGNE)) {
      _mqttConsigne = payload.toFloat(); //_mqttConsigne is a float
    }
  });
}
```

The compiler becomes very gross at this point:
```
Network.cpp:159: error: 'this' was not captured for this lambda function
       _mqttConsigne = payload.toFloat();
       ^
```

I spent a long time asking Google about this and it reveals that my callback function is static, so it won't work. I couldn't find a way to assign the payload to a member variable, mainly because it goes far beyond my understanding.
Would you show any piece of code of your library included inside a class, or am I just trying to do that the wrong way?

256dpi, 2017-11-15:
You need to capture the implicit `this` variables as `_mqttConsigne` actually evaluates to `this._mqttConsigne`: https://crascit.com/2015/03/01/lambdas-for-lunch/

This should do the job:

```c++
void Network::beginMqtt() {
  _mqtt.setWill(MQTT_TOPIC_WILL,"offline",true,2);
  _mqtt.begin("jeedom.vt.lan", _wifiClient);
  _mqtt.onMessage([this](String &topic, String &payload) {
    Serial.println(topic + ": " + payload);
    if (topic.equals(MQTT_TOPIC_SET_CONSIGNE)) {
      this._mqttConsigne = payload.toFloat(); //_mqttConsigne is a float
    }
  });
}
```

When the lambda is created and passed to `_mqtt.onMessage()' the `this` of the current `Network` instance is captured and later used inside the callback.

mathieu-mp, 2017-11-15:
Thank you for your time.

I liked the suggestion much, looked pretty straight:
```cpp
void Network::beginMqtt() {
  _mqtt.setWill(MQTT_TOPIC_WILL,"offline",true,2);
  _mqtt.begin("jeedom.vt.lan", _wifiClient);
  _mqtt.onMessage([this](String &topic, String &payload) {
    Serial.println(topic + ": " + payload);
    if (topic.equals(MQTT_TOPIC_SET_CONSIGNE)) {
      this._mqttConsigne = payload.toFloat(); //_mqttConsigne is a float
      Serial.println("Consigne Set");
    }
  });
}
```
But it ended badly:
```
Arduino: 1.8.4 (Windows 10), Board: "WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"
sketch\Network.cpp: In lambda function:
Network.cpp:159: error: request for member '_mqttConsigne' in 'this', which is of pointer type 'Network*' (maybe you meant to use '->' ?)
       this._mqttConsigne = payload.toFloat();
            ^
sketch\Network.cpp: In member function 'void Network::beginMqtt()':
Network.cpp:162: error: no matching function for call to 'MQTTClient::onMessage(Network::beginMqtt()::__lambda5)'
   });
    ^
sketch\Network.cpp:162:4: note: candidate is:
In file included from sketch\Network.h:22:0,
                 from sketch\Network.cpp:10:
C:\Users\Mathieu Payrol\Documents\Bricolages\Arduino\Arduino_Sketchbook_location\libraries\MQTTClient\src/MQTTClient.h:118:8: note: void MQTTClient::onMessage(MQTTClientCallbackSimple)
   void onMessage(MQTTClientCallbackSimple cb) {
        ^
C:\Users\Mathieu Payrol\Documents\Bricolages\Arduino\Arduino_Sketchbook_location\libraries\MQTTClient\src/MQTTClient.h:118:8: note:   no known conversion for argument 1 from 'Network::beginMqtt()::__lambda5' to 'MQTTClientCallbackSimple {aka void (*)(String&, String&)}'
exit status 1
request for member '_mqttConsigne' in 'this', which is of pointer type 'Network*' (maybe you meant to use '->' ?)
```

OK, so let's try what's suggested (at this point, I don't really understand what I'm doing):
```c++
void Network::beginMqtt() {
  _mqtt.setWill(MQTT_TOPIC_WILL,"offline",true,2);
  _mqtt.begin("jeedom.vt.lan", _wifiClient);
  _mqtt.onMessage([this](String &topic, String &payload) {
    Serial.println(topic + ": " + payload);
    if (topic.equals(MQTT_TOPIC_SET_CONSIGNE)) {
      this->_mqttConsigne = payload.toFloat(); //_mqttConsigne is a float
      Serial.println("Consigne Set");
    }
  });
}
```
And there we go again:
```
Arduino: 1.8.4 (Windows 10), Board: "WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"
sketch\Network.cpp: In member function 'void Network::beginMqtt()':
Network.cpp:162: error: no matching function for call to 'MQTTClient::onMessage(Network::beginMqtt()::__lambda5)'
   });
    ^
sketch\Network.cpp:162:4: note: candidate is:
In file included from sketch\Network.h:22:0,
                 from sketch\Network.cpp:10:
C:\Users\Mathieu Payrol\Documents\Bricolages\Arduino\Arduino_Sketchbook_location\libraries\MQTTClient\src/MQTTClient.h:118:8: note: void MQTTClient::onMessage(MQTTClientCallbackSimple)
   void onMessage(MQTTClientCallbackSimple cb) {
        ^
C:\Users\Mathieu Payrol\Documents\Bricolages\Arduino\Arduino_Sketchbook_location\libraries\MQTTClient\src/MQTTClient.h:118:8: note:   no known conversion for argument 1 from 'Network::beginMqtt()::__lambda5' to 'MQTTClientCallbackSimple {aka void (*)(String&, String&)}'
exit status 1
no matching function for call to 'MQTTClient::onMessage(Network::beginMqtt()::__lambda5)'
```
I'm wondering if onMessage is able to handle that syntax.
Any other suggestion?

Thanks again.

256dpi, 2017-11-15:
Yes the replacement with a `->`  is correct as `this` is a pointer.

I also just learned that as soon you capture a variable the lambda cannot be used as a function pointer anymore: https://stackoverflow.com/questions/28746744/passing-lambda-as-function-pointer

Can't you just assign the result to a global variable?

mathieu-mp, 2017-11-17:
As you suggested, I stopped boiling my brain and ended filling a static fifo variable with mqtt packets. This fifo is declared as static inside my `class.h` and initialized inside my `class.cpp`. As soon as possible (back into my object inside `Network.loop()`), I empty the fifo triggering the actions based on the topic.

In case it helps anybody here is the code, please note I used a fifo template class referenced on arduino.cc
Network.h
```cpp
struct MqttInput {
  String topic;
  String payload;
};

class Network
{
  public:
    /* ... */
    void loop();
  private:
    void mqttBegin();
    static QueueList <MqttInput> _mqttInputQueue;
};
```
Network.cpp
```cpp Network.h
QueueList <MqttInput> Network::_mqttInputQueue;

void Network::mqttBegin() {
  _mqtt.setWill(MQTT_TOPIC_TX_WILL,"offline",true,2);
  _mqtt.begin(MQTT_BROKER, _wifiClient);
  _mqtt.onMessage([](String &topic, String &payload) {
    _mqttInputQueue.push({topic, payload});
  });
}

void Network::loop() {
  /* ... */
  while(!_mqttInputQueue.isEmpty()) {
    MqttInput mqttInput = _mqttInputQueue.pop();
    if ((mqttInput.payload.length() > 0) && (mqttInput.topic.length() > 0)) {
      if (mqttInput.topic.equals(MY_TOPIC_NAME)) {
        _heaterControl.setConsigne(mqttInput.payload.toFloat());
      }
    }
  }
}
```
@256dpi, thank you very much about your help! (Where is the beer link?)
And again: great arduino mqtt library!

closed 2017-11-17 by mathieu-mp

256dpi, 2017-11-17:
This FIFP approach is anyway better, since the callback can also be triggered during a call to `subscribe`, `unsubscribe` or `publish` with QoS > 0.

FYI: You can always use the low level lwmtt lib to create custom client, if you have special needs. ;)