256dpi/arduino-mqtt #86: Subscribing issue

giacomopastore, 2017-11-24:
Hi Joël,

I've tried to implements a subscribing function on my sketch but when I send messages from external application I can't see anything on the other side. I'm started from your example commenting publish method and replacing MQTT address ("arduino") with a real MQTT server IP.

Also I've tried your example and I've noticed that If I change the IP address of the broker (using a fake IP instead "arduino" or my real MQTT broker) I can receive messages. 

How can I do?
Thanks very much

Jack

256dpi, 2017-11-27:
Can you attach your code?

giacomopastore, 2017-11-27:
Hi Joël,

problem fixed! This morning I checked my code and now seems works right.
Sorry if I inserted the issue early.

Jack

closed 2017-11-27 by giacomopastore