256dpi/arduino-mqtt #90: User with NULL password cannot connect

robcazzaro, 2017-12-23:
I'm connecting to a MQTT broker that requires a user name but null password. In your MQTTClient.h header file you have the following code that only sets a user name if the password is not null

    if (username != nullptr && password != nullptr) {
      options.username = lwmqtt_string(username);
      options.password = lwmqtt_string(password);
    }

I suggest modifying it so that a user name with null password can be accepted

256dpi, 2017-12-25:
Thanks for reporting this! I fixed it and released a new version (2.2.2). It will take some time until it appears in the Arduino library manager. Merry Christmas!

robcazzaro, 2017-12-26:
Merry Christmas to you too! I'm glad to have been of help, I'm a very happy user of your library

closed 2017-12-26 by robcazzaro