256dpi/arduino-mqtt #91: onMessage callback not called on ESP32

alexfederlin, 2017-12-26:
I am using an nodeMCU ESP32 to run the below code. It appears to be working fine, MQTT messages get published, but messages to the subscribed topic do not trigger the callback.
The server does send MQTT Publish Messages to the IP of the device, but nothing is written to the terminal...

`#include <Arduino.h>
#include "WiFi.h"
#include <MQTTClient.h>
#include <SPI.h>

WiFiClient net;
MQTTClient client;
unsigned long lastMillis = 0;

// WiFi credentials.
const char* WIFI_SSID = "iot";
const char* WIFI_PASS = "iot123456";

void messageReceived(String &topic, String &payload) {
  Serial.println("incoming: " + topic + " - " + payload);
}

void connect() {
  Serial.print("checking wifi...");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(1000);
  }

  Serial.print("\nconnecting to MQTT Broker...");
  while (!client.connect("arduino", "try", "try")) {
    Serial.print(".");
    delay(1000);
  }

  Serial.println("\nconnected!");

  client.subscribe("/test/#");
  // client.unsubscribe("/hello");
}

void setup()
{
    Serial.begin(9600);
    // Giving it a little time because the serial monitor doesn't
    // immediately attach. Want the firmware that's running to
    // appear on each upload.
    delay(2000);

    Serial.println();
    Serial.println("Running Firmware.");

    // Connect to Wifi.
    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(WIFI_SSID);

    // Set WiFi to station mode and disconnect from an AP if it was previously connected
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(100);

    WiFi.begin(WIFI_SSID, WIFI_PASS);
    Serial.println("Connecting to Wifi...");

    while (WiFi.status() != WL_CONNECTED) {
      // Check to see if connecting failed.
      // This is due to incorrect credentials
      if (WiFi.status() == WL_CONNECT_FAILED) {
        Serial.println("Failed to connect to WIFI. Please verify credentials: ");
        Serial.println();
        Serial.print("SSID: ");
        Serial.println(WIFI_SSID);
        Serial.print("Password: ");
        Serial.println(WIFI_PASS);
        Serial.println();
      }
      delay(5000);
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());

    Serial.println("Hello World, I'm connected to the internets!!");

    // Note: Local domain names (e.g. "Computer.local" on OSX) are not supported by Arduino.
    // You need to set the IP address directly.
//    client.begin("broker.shiftr.io", net);
    client.begin("192.168.2.2", net);
    client.onMessage(messageReceived);
    connect();

}

void loop()
{
    delay(5000);
    Serial.println("Hello World");

    if (!client.connected()) {
      connect();
    }

    // publish a message roughly every second.
    if (millis() - lastMillis > 1000) {
      lastMillis = millis();
      client.publish("/hello", "world");
    }
}
`

256dpi, 2017-12-27:
From what topic do you expect to receive messages? In your code you publish to "/hello" and subscribe to "/test/#".

alexfederlin, 2017-12-27:
Sorry, should have been more specific. I was using MQTTBox to publish on
topic "/test"

256dpi, 2017-12-27:
Did you try a topic without a wildcard? Maybe the broker is messing stuff... I don't think that this is a library bug as receiving a MQTT message is a key functionality and it would be quite weird if that would not work suddenly. ;)

alexfederlin, 2017-12-27:
Yes, same thing when subscribing to plain topic "/test"
I would indeed be surprised if there is anything at fault with the library. This was more to check if I used it correctly or if missed something.

256dpi, 2017-12-28:
The code looks fine to me, can you test it with shiftr.io? Maybe the MQTTBox broker is messing up.

alexfederlin, 2018-01-01:
I also checked with shiftr.io. The messages seem to be sent to the device just fine (already confirmed this using wireshark with my local MQTT setup as well). 
I am wondering if the wifi connection or wifi library on the device are somehow messed up (or in some strange way incompatible with the cheap China knock-off board I'm using). But in addition to my local AP, I also connected the device to my phone's hotspot with the same effect.
Is there any way to check for incoming packets on the network layer on the ESP32 device?

256dpi, 2018-01-10:
I hade time to look at your example a little bit more in detail. It turns out that the bug is a missing call to `client.loop()` in `void loop {}`. Without this call the client will not read from the network and receive incoming messages. Also, you should never use `delay(x)` if you use MQTT as it can prevent the client from receiving data or keep the connection alive.

closed 2018-02-08 by 256dpi

alexfederlin, 2018-02-09:
Sorry I haven't responded earlier. Just saw your last comment which makes sense. Will test out at the next opportunity. Thanks for your support!

lhanson, 2019-10-07:
> Also, you should never use `delay(x)` if you use MQTT as it can prevent the client from receiving data or keep the connection alive.

@256dpi I have a device where I only want to publish a sensor reading every 15 minutes and receive a subscription to look for any new incoming configuration changes. The way I'm currently doing that is to `delay(x)` and then reconnect, and I'm finding that I don't get forwarded messages even when they're sent as retained (even when I'm able to reconnect and send just fine). Your comment makes me think that perhaps this is expected; how would you recommend sleeping for 15 minutes without losing messages?