256dpi/arduino-mqtt #96: LWMQTT_BUFFER_TOO_SHORT though messages are well shorter than 128 bytes

mkarlesky, 2018-02-02:
First of all, thank you for this library. The hard work invested in it is evident, and I'm eager to get it working as MQTT seems a perfect fit for a certain project.

I can't seem to get past an error. When working with mosquitto (1.4.12) I see the logging messages quoted below repeating forever, and my arduino code tries to connect to the broker over and over again. I've tried two different local brokers (mosquitto and emqttd) plus an online broker and seemingly have the same issue with all three. When I get errors back from `lastError()` they're always `LWMQTT_BUFFER_TOO_SHORT` though clearly the messages involved will fit within the default sized buffer with room to spare. I've tried increasing the buffer to 256 bytes with no change.

I'm stumped on what I might be doing wrong or if there's an issue with `EthernetClient`. To my knowledge my Ethernet2 library and hardware is working properly (at least for UDP elsewhere in the code base).

I'm working with a Mega2560 with an Ethernet2 shield and the Ethernet2 library. I'm not using the Arduino IDE. Instead I'm working with a build environment that calls avr-gcc directly and configures search paths, compilation, and linked libraries. I'm using arduino-mqtt 2.2.2 and Ethernet2 1.0.4 (latest releases of each).

The code below is some very simple prototyping based on the provided arduino-mqtt examples. In the class structure of my application, Executors have `setup()` and `upkeep()` methods that are equivalent to an Arduino sketch's `setup()` and `loop()`.

Any guidance on debugging would be most appreciated. It seems quite unlikely to be a problem with the arduino-mqtt library itself.

```
1517531809: New connection from 172.17.0.1 on port 1883.
1517531809: New client connected from 172.17.0.1 as Tester (c1, k10).
1517531809: Sending CONNACK to FarmHand (0, 0)
1517531810: Socket error on client FarmHand, disconnecting.
```
---
```
MqttExecutor::MqttExecutor(
  ISuperExecutor& superExecutor,
  ITimerFactory& timerFactory,
  ILogger& logger) :
  Executor(superExecutor, 0),
  _timerFactory(timerFactory),
  _timer(nullptr),
  _logger(logger),
  _eth(*(new EthernetClient())),
  _client(*(new MQTTClient()))
{
  // Do nothing
}

bool MqttExecutor::setup() {
  bool success = true;
  
  _client.begin("10.0.1.99", _eth);

  success &= ((_timer = _timerFactory.manufacture(1000)) != nullptr);

  while(!_client.connected()) {
    _client.connect("Tester");
    delay(500);
  }
  
  return success;
};

void MqttExecutor::upkeep() {  
  lwmqtt_err_t err;

  _client.loop();

  err = _client.lastError();
  if (err != LWMQTT_SUCCESS) {
    SPRINTF_LOG_STRING(message, "MQTT error: %d", err);
    LOG_ERROR(_logger, message);
  }
  
  if (!_client.connected()) {
    if (!_client.connect("Tester")) {
      LOG(_logger, "MQTT: Could not connect");
      return;
    }
    LOG(_logger, "MQTT: Connected");    

    err = _client.lastError();
    if (err != LWMQTT_SUCCESS) {
      SPRINTF_LOG_STRING(message, "MQTT error: %d", err);
      LOG_ERROR(_logger, message);
    }
  }
    
  if (_timer->isExpired()) {
    _client.publish("/hello", "world");
    _timer->reset();
  }
}
```

256dpi, 2018-02-02:
Where is `_client` initialized? Receive and send buffers are allocated within the constructor. Is it possible that the object is not properly initialized?

256dpi, 2018-02-02:
Nevemind, I found it. Which method exactly does return the `LWMQTT_BUFFER_TOO_SHORT` error?

mkarlesky, 2018-02-02:
The error `LWMQTT_BUFFER_TOO_SHORT` is always logged by the first call to `_client.lastError()`, immediately after `_client.loop()`. However, it seems to oscillate between the error and `LWMQTT_SUCCESS` in a pattern corresponding to the connect / disconnect / reconnect behavior I'm seeing.

256dpi, 2018-02-02:
So the only thing I can imagine is that the broker is trying to send a message that is too long. Maybe an older published message that somehow is inflight?

mkarlesky, 2018-02-02:
But the peculiar thing is that in my setup there's only one client publishing (my arduino) and one thing subscribing (a command line MQTT client). I'm admittedly new to MQTT, but I can't think of where a long message would be coming from. I can say that everything behaves as expected when the broker shuts down and restarts. I see failed connection attempts, etc. I assume that restarting the broker would flush any inflight messages.

My current hunch is that something in my setup is somehow causing erroneous string buffer processing that is then triggering misleading error states.

Do you have a preferred setup for lwmqtt debugging? My next step is to dig in to find out exactly where `LWMQTT_BUFFER_TOO_SHORT` is emitted.

Thanks for the quick responses and the support. It's very appreciated.

mkarlesky, 2018-02-02:
My hunch was correct. The `message` struct parameter is garbled once inside `lwmqtt_publish()`; `payload_len` is a ridiculous value. Coincidentally, this is the boundary between our library build of `arduino-mqtt` and the executable build of our project that links it. I'm looking into byte packing and struct alignment compilation issues.

mkarlesky, 2018-02-02:
Bingo. `-fshort-enums` was present in our project build but not in our libraries build. Apparently, arduino-mqtt is the only library we're using that includes structs that in turn contain enums. The disparity was leading to byte misalignment that corrupted message length handling. Now that the builds have been brought to parity, everything runs as expected!

Thank you again for your responsiveness and support.

closed 2018-02-02 by mkarlesky

256dpi, 2018-02-04:
I'm glad to hear you found a solution!