256dpi/arduino-mqtt #101: Mqtt connection errors?

tyczj, 2018-03-02:
Just curious if there was a way to find out if a connection fails what was the reason it failed?

256dpi, 2018-03-03:
You can use:

```c++
lwmqtt_err_t lastError();
lwmqtt_return_code_t returnCode();
```

The error values can be found here: https://github.com/256dpi/lwmqtt/blob/master/include/lwmqtt.h#L11.
And the return codes here: https://github.com/256dpi/lwmqtt/blob/master/include/lwmqtt.h#L243

tyczj, 2018-03-03:
Ah nice, so I am getting `LWMQTT_BUFFER_TOO_SHORT` what does that mean I have to do?

256dpi, 2018-03-03:
> The maximum size for packets being published and received is set by default to 128 bytes. To change the buffer sizes, you need to use MQTTClient client(256) instead of just MQTTClient client on the top of your sketch. The passed value denotes the read and write buffer size.

Please consider reading the README before asking any further questions.

closed 2018-03-03 by tyczj