256dpi/arduino-mqtt #103: Error 246 (get last error) when trying to connect ...

euquiq, 2018-03-14:
Hi ! 

I 've been trying using this library on my Arduino ESP32 project. 

When connecting into flespi mqtt broker (mqtt.flespi.io)  I get Error 246.

So I am not able to establish a connection. I tried both ports 1883 and 8883, with either WiFiClient and WiFiClientSecure.

When I replace your library with PubSubClient library (and change the different function names) It works fine. 

I would like to use your library because when I use PubSubClient, under WiFiClientSecure, whenever y try to perform a PubSubClient.disconnect()  my ESP32 resets itself (Grrr!) and I noticed that your library seems to be more active and up to date, so I was hoping to get it working OK.

Any insight or ideas are welcome!

Euquiq

256dpi, 2018-03-15:
Can you share some code?

euquiq, 2018-03-15:
Sorry, I can't, I did return to PubSubClient, and found a way to fix my original error in it. 

I wonder from where (and what eventual meaning) would have those error codes your (and also PubSubClient) library returns when there is a problem: Probably it should be obvious but I don't know about it, nor could I find an errors list anywhere. I will close this issue, now.

closed 2018-03-15 by euquiq