256dpi/arduino-mqtt #112: Last Will is triggered by client reconnecting after deep sleep

reibuehl, 2018-06-25:
I connect to MQTT with a last will set to payload "Offline"and a keepAlive timeout of 80 sec. My code publishes "Online" to the same topic as the last will and then takes less than 2 sec for other stuff.  Then the ESP8266 goes to deep sleep for 60 seconds. When it comes back from deep sleep and reconnects to MQTT, this triggers first the last will message "Offline" being send out by the broker and a few milliseconds later the "Online" comes from my code. I already tried to increase the keepAlive from 65 to 80 seconds since I thought it is a race condition and tried setting clearSession to false but both seem to have no effect on this behavior. 
Is there a way to prevent the last will being sent out on reconnect? I only want it to be send by the broker if the client does not connect for more than 65 seconds aka misses its regular interval.

256dpi, 2018-06-25:
This is a very advanced use case. I have no idea what happens to TCP connections when going to deep sleep. I assume the MQTT client will reconnect because the TCP connection is flagged as broken. Can you share an example?

reibuehl, 2018-06-28:
I will try if I can isolate the MQTT part from the rest of my code to show the behavior.

reibuehl, 2018-06-29:
I tried creating an example but the result is not showing the same behavior as my full code. With the sample code, the last will is send out immediately after the ESP goes to deep sleep. That is more logical than the behavior I see with my code but unfortunately also not the way I was hoping it would work but this might be a limitation of the MQTT protocol itself.
I still attach the sample code here in case somebody has an idea:
```
// This example uses an Wemos D1 mini ESP8266 board
// to connect to a MQTT server and uses deep sleep.
// 
// Connect D0 <-> RST and A0 <-> 3V3 with jumper cables
// Remove the jumper cable from RST during flashing.
//
// by Reiner Bühl, <reiner@buehl.net>

#include <ESP8266WiFi.h>
#include <MQTTClient.h>

// Compare VCC against internal reference voltage
ADC_MODE(ADC_VCC); //vcc read

// minimum Battery voltage
float vBatMin = 3.0;

const char ssid[] = "<WIFI SSID>";
const char pass[] = "<WIFI PASSWORD>";
const char mqttserver[] = "<MQTT SERVER>";

WiFiClient net;
MQTTClient client;

void connect();  // <- predefine connect() for setup()

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println();
  Serial.println("Setup...");

  WiFi.begin(ssid, pass);

  // Note: Local domain names (e.g. "Computer.local" on OSX) are not supported by Arduino.
  // You need to set the IP address directly.
  client.begin(mqttserver, net);
  client.onMessage(messageReceived);
  client.setOptions(80, false, 1000); // set keepAlive to 80 seconds (default 10 seconds)
                                      // and clearSession to false
  client.setWill("/SmartHome/D1mini/state", "Offline", true, 2);

  connect();

  client.publish("/SmartHome/D1mini/state", "Online");

  Serial.println("Setup completed...");
}

void connect() {
  Serial.print("checking wifi...");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(1000);
  }

  Serial.print("\nconnecting...");
  while (!client.connect("arduino", "try", "try")) {
    Serial.print(".");
    delay(1000);
  }

  Serial.println("\nconnected!");
}

void loop() {
  if (!client.connected()) {
    connect();
  }

  // Check if battery level is good
  float vBat = ESP.getVcc() / 1024.0;
  static char vBatVoltage[10];
  dtostrf(vBat, 5, 3, vBatVoltage);
  sprintf(vBatVoltage,"%s V", vBatVoltage);
  Serial.println(vBatVoltage);  
  client.publish("/SmartHome/D1mini/batteryVoltage", vBatVoltage, false, 2);
  static char vBatLevelOk[3] = "ok"; 
  static char vBatLevelLow[4] = "low"; 
  if ( vBat >= vBatMin ) {
    Serial.println("Publish vBatLevelOk");  
    client.publish("/SmartHome/D1mini/batteryState", vBatLevelOk, false, 2);
  } else {
    Serial.println("Publish vBatLevelLow");  
    client.publish("/SmartHome/D1mini/batteryState", vBatLevelLow, false, 2);
  }
  
  // send publishs
  client.loop();
  delay(10);  // <- fixes some issues with WiFi stability
  Serial.println("Client loop done");  

  // Deep sleep mode for 30 seconds, the ESP8266 wakes up by itself when GPIO 16 (D0 in NodeMCU board) is connected to the RESET pin
  Serial.println("Going into deep sleep mode for 60 seconds");
  ESP.deepSleep(60e6);
}

void messageReceived(String &topic, String &payload) {
  Serial.println("incoming: " + topic + " - " + payload);
}

```

TotallyInformation, 2018-09-01:
The LWT message should be sent by the Broker and not the client. It is set by the client connect but shouldn't be triggered by it. Nor by a client disconnect. My understanding is that the LWT message should only be triggered by an unexpected client loss which, by definition, should happen after the broker times out the connection (by not having received a message quickly enough).

> The broker stores the message until it detects that the client has disconnected ungracefully. In response to the ungraceful disconnect, the broker sends the last-will message to all subscribed clients of the last-will message topic. If the client disconnects gracefully with a correct DISCONNECT message, the broker discards the stored LWT message.

[From HiveMQ](https://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament)

Couple of thoughts - firstly, what broker are you using? Secondly, is the above code written to more than 1 device? You are connecting with a fixed client name I think ("arduino") and this will cause issues if you have >1 connection.

reibuehl, 2018-09-01:
I used Mosquitto as broker and unique client names - the "arduino" is just from the sample code, I use different settings in my environment.

Urs-Eppenberger, 2018-10-01:
I'm in very deep water here. But I played around with the keepAlive and timeout values too.
There are some hints on the web about good combinations.
The one you have is a keepAlive of 80 seconds and a timeout of 1 second.
I suggest you experiment with the timeout value.
In my case I have set it to 25 seconds, and my sensors send a message every 10 seconds. That way I got quite stable connections.
On another note to @Joël : I first used the setOptions command. It had no effect, strangely. Then I changed the value directly in your code in the file MQTTClient.h  Then it worked. I'm not a programmer, and I did not really thoroughly check and verify the behaviour. I would have to listen on the wire for the packets and check what exactly happens between my mosquitto broker on the pi and the esp8266 based clients.

256dpi, 2019-02-08:
This is most likely related to #144 and will also be improved by #141. I'd like to move the discussion over there. If there are issues after these things have been fixed. Please reopen or open a new issue.

closed 2019-02-08 by 256dpi