256dpi/arduino-mqtt #119: subscribe to two different topics

Urs-Eppenberger, 2018-08-06:
Hello Joël

Is it possible to subscribe to two different topics?
In my case I would need that for the following scenario:
- command/sensor09  for commands
- cellar/temp  for temperature values from another sensor
I have an actor which receives commands on one topic and fetches temperature values from another topic

And if it is possible to subscribe to two topics, is it also possible to have specific callback functions, depending on the subscribed topic? If yes, I'm in desperate need of some simple example code, since it is not obvious from the examples.

Kind regards,

Urs.

Urs-Eppenberger, 2018-08-07:
I found something which works. Should have googled before asking.
https://www.baldengineer.com/multiple-mqtt-topics-pubsubclient.html?utm_campaign=coschedule&utm_source=tumblr&utm_medium=baldengineer&utm_content=Multiple%20MQTT%20Topics%20with%20Arduino%20PubSubClient

closed 2018-08-07 by Urs-Eppenberger

256dpi, 2018-08-07:
Here is an alternative that uses Arduino's `String` object:

```c++
void messageReceived(String &topic, String &payload) {
  if(topic.equals("command/sensor09")) {
    // ...
  } else {
    Serial.println("incoming: " + topic + " - " + payload);
  }  
}
```