256dpi/arduino-mqtt #122: Basic example for onMessageAdvanced

FernandoGarcia, 2018-09-09:
Hi!

Can you give me an example of usage for onMessageAdvanced please?

I'm trying the method used in others libraries but i'm getting compiling errors.

my code is like this:

`_mqttClient.onMessageAdvanced(callback);`

I tried:

```
void callback(String topic, String payload, char * bytes, unsigned int length){}

void callback(MQTTClient *client, char topic[], char payload[], int payload_length){}

void callback(_client, char topic[], char payload[], int payload_length){}
```

But always getting this type of error.

> error: invalid conversion from 'void (*)(char*, byte*, unsigned int) {aka void (*)(char*, unsigned char*, unsigned int)}' to 
> 'MQTTClientCallbackAdvanced {aka void (*)(MQTTClient*, char*, char*, int)}' [-fpermissive]

In example folder there's only usage of simple callback like:

`void messageReceived(String &topic, String &payload) {}`

Sorry for this noob's question.

Best regards.

256dpi, 2018-09-11:
Sure here is the correct implementation: 

```c++
void messageReceived(MQTTClient *client, char topic[], char payload[], int payload_length) {
  Serial.println(String("incoming: ") + String(topic) + " - " + String(payload_length));
}
```

FernandoGarcia, 2018-09-11:
Hi!

Thanks for your answer!

It was my mistake because I have declared function prototype in a way and function in another.

As I'm moving the code from Pubsubclient I forgot to change the protype in other file (global variables).

It was in a file:

`void callback(char topic[], char payload[], int payload_length);`

And another:

`void callback(MQTTClient *client, char topic[], char payload[], int payload_length)
{
}`

Thanks for your help!

Best regards.

closed 2018-09-11 by FernandoGarcia