256dpi/arduino-mqtt #123: Consider performing deep copy of LWT or passing it as part of the connect command

kubark42, 2018-09-17:
The code shows that the last will and testament is not making a deep copy, but is instead setting a pointer to the strings containing the topic and the message. These two strings are not guaranteed to be long-lived, and so their memory is quite possibly freed-- and overwritten-- by the time the connect command has been called.

C.f.: https://github.com/256dpi/arduino-mqtt/blob/61a253d4c5edec3e27f471e172df33d6f1039818/src/MQTTClient.h#L152

It would be good to pass the messages as arguments during the `connect()` command, or to rewrite so that the LWT is permanently stored.

I don't have a sense as to which of the two solutions is better. The first lacks in syntactic beauty, the second becomes inefficient. Since LWMQTT is designed to have a minimal footprint, it might be that the RAM cost of permanently storing the LWT isn't reasonable.

256dpi, 2018-09-28:
Yes, I think you're right. I will start working on a patch.

kubark42, 2018-10-11:
Cool beans!