256dpi/arduino-mqtt #124: disconnect() cause Exception (28)

coelner, 2018-09-26:
This code snipper cause an exception
```
bool mqtt_disconnect() {
  mqttClient.loop();
  delay(50);
  bool retVal = mqttClient.disconnect();
  delay(10);
#ifdef SERIAL_DEBUG
  if ( retVal ) {
    Serial.println("mqtt disconnected");
  }
  else {
    Serial.println("mqtt NOT disconnected");
  }
#endif
  return retVal;
}
```

```
Exception (28):
epc1=0x402064b7 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000064 depc=0x00000000

ctx: cont 
sp: 3ffffd20 end: 3fffffc0 offset: 01a0

Decoding 13 results
0x40104b6e: ets_timer_setfn at ?? line ?
0x40104d40: ets_timer_arm_new at ?? line ?
0x40208c80: esp_yield at Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 91
0x402025e6: delay at Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.c line 54
0x402064b5: ClientContext::wait_until_sent(int) at Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 196
0x40203ae8: lwmqtt_send_packet_in_buffer at Arduino/libraries/arduino-mqtt/src/lwmqtt/client.c line 186
0x4020651d: WiFiClient::flush(unsigned int) at Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 196
0x40206479: WiFiClient::stop(unsigned int) at Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 196
0x40208f11: MQTTClient::close() at Arduino/libraries/arduino-mqtt/src/MQTTClient.h line 383
0x4020525d: MQTTClient::disconnect() at Arduino/esp8266-test/ota.ino line 66
:  (inlined by) mqtt_disconnect() at Arduino/esp8266-test/mqtt.ino line 76
0x40205880: setup at Arduino/esp8266-test/esp8266-test.ino line 678
0x40208d04: loop_wrapper at Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 122
0x40100739: cont_wrapper at Arduino/hardware/esp8266com/esp8266/cores/esp8266/cont.S line 81
```

As soon as I skip the `mqttClient.disconnect()` the sketch runs fine.

256dpi, 2018-09-28:
Please post the complete sketch. I have the feeling other side effects cause the issue.

coelner, 2018-10-01:
[esp8266-test.zip](https://github.com/256dpi/arduino-mqtt/files/2432775/esp8266-test.zip)

coelner, 2018-10-02:
reflashed the sketch with the current commit (24a1d22) and it is working flawless so far

closed 2018-10-02 by coelner