256dpi/arduino-mqtt #128: Jwt connection

Gior80, 2018-10-24:
Hi,
Google iot core uses jwt as connection credential . Would you have an example of how to implement it ?

256dpi, 2018-10-25:
AFAIK you should be able to submit the JWT as the username/password  when connecting. MQTT 3.1.1 does not offer other means to transmit this information.

Gior80, 2018-10-25:
thanks for your answer Joel,  I'm sorry but maybe I've not exposed this thing good and you are maybe expert than me in suggesting a right approach. 

Actually I'm able to connecting Jwt,  it works perfect.  Point is only when jwt expires.  The expiring is set via server side ( actually for google is about 1 hour). Standing at google doc it's possible to renew without disconnect: 
https://cloud.google.com/iot/docs/how-tos/credentials/jwts#refreshing_jwts

only way I found to renew the connection with pubsub is to disconnect the client and reconnect it with new credential. this gonna break my connection for many seconds that I cannot miss.   there's should be a way to say server "hey this is my new token, please use it" 
just feel absolutely free to tell me if I'm missing something here.....

Gior80, 2018-10-28:
hi Joel, 

tested as per your instruction.   I'm unable to get a connection
.-10
4
.-10
5

as I was using another library for generating USR and pass I'm quite sure is is working ok. btw what I would like to ask you: 
as per google iot core requirement Username and password are SO long.  Is there any limitation in device and password length ?   this is what raised to my mind immediately 

ex.: device:
projects/myproject/locations/europe-west1/registries/myreg/devices/testdevice

JWT Password example: eyJhbGciOiJFUzI1NiIewrWrwecCI6IkpXVCJ9.eyJpYXQiOjAsImV4cCI6MzYwMCwiYXVkIjoiY2hhcmdlc3BsIn0=.mDxK8R1234567zk06MeWYAxXqXsJBKvjwoMqpXfV6M4Jq6xfLpqkmMACSPNaNRc4jGUHVucpKFRXVVZGw3izlQ==

Gior80, 2018-11-05:
This issue is solved , can be closed

256dpi, 2018-11-05:
What was the solution?

closed 2018-11-05 by 256dpi

Gior80, 2018-11-05:
sorry forgot to write,  solution was in payload size