256dpi/arduino-mqtt #129: Possible to make it work on RasPiArduino?

Barabba11, 2018-10-24:
Hi there, thank you for your help.
This is not properly an issue, but I don't know how to ask help in public way, I hope someone can help and maybe someone else can find info useful in the future.

I'm using Arduino IDE to create a file .bin for my raspberry as a "driver" for the I2C hrdware used for my domotic (reley, sensors, etc.), I need to communicate with Home Assistant by MQTT, I've tested all the hardware already, the last thing now is to include an MQTT client into the .bin file.

I really hope the RasPiArduino deals with the Raspberry ethernet port, I just need to interface the IDE with that, do you think your script can work there? If I copy/past your example code I obtain an error of compiling, probably because I can't deal with Raspberry's Ethernet like a shield, there is already an IP address and so on, how to use it? Thank you a lot

Barabba11, 2018-10-25:
Thank you a lot for your kind answer! You can't imagine how precious it is for me! I can't imagine to learn another programming language like phyton in so short time if RasPiArduino can't compile MQTT.. so you really save me. I'll test it as soon as possible.
By the way do you mean this?
https://github.com/bobbytech/Easy-Arduino-Yun-MQTT-Client
Looks easy and powerful in the example, I've just to understand more this dependency:

_"A_ mosquitto client on the lininio side(openWRT) is needed, and can be installed using the package manager (http://arduino.cc/en/Tutorial/YunPackageManager). Just run these two commands on your root@yun.local to _install___

If the code is wrote for Arduino, what sense it has to execute from Linux? What Linux on Arduino? 
Thank you

Barabba11, 2018-10-25:
How can I use the "bridge library", will it be for me just necessary to 
#include <Process.h>
#include "MQTTclient.h"
like in the example or shoudl I do something else? 
Thanks a lot!

Barabba11, 2018-10-31:
caring about the libraries (it was using original Ethernet.h intstead of Raspiarduino one I made it compiling
https://github.com/bobbytech/Easy-Arduino-Yun-MQTT-Client/blob/master/examples/MQTTclient_example/MQTTclient_example.ino
but now I have a "killed" when I start from SSH. I think the bridge is not working

Barabba11, 2018-10-31:
hi mate, tried another one that doesn't create problems with SSH session or being "killed", the 
Anyway it doesn't connect, any IP packet coming from Raspberry, when you wrote "RasPiArduino readme and found: "Process, FileIO, Client, Server" cient server and UDP implementations through the Bridge library" client and server means TCP?

Barabba11, 2018-10-31:
I had success with Adafruit_MQTT_Library, Raspberry will take a new IP address from DHCP for the bridge, and still keep the static one I assigned for SSH

256dpi, 2019-02-08:
I'm closing this, as it is more of a general question.

closed 2019-02-08 by 256dpi