256dpi/arduino-mqtt #130: Strange behavior of MKR GSM 1400

gamelaster, 2018-10-31:
Hi,
I have MKR GSM 1400, I have verified that GPRS cellular data works.

When I'm testing the MKR GSM 1400 example (no secure), everything works well
When I'm testing the MKR GSM 1400 example (secure), it can't connect (last Error: -6, error Code: 6) (It's doing the same for Adafruit IO 8883).
When I change MRK GSM 1400 example (no secure), to send data to my Adafruit IO, it connects, but in loop, the `connected` function still returns false, so it's still reconnecting. Anyway, when I comment this condition, it successfully sends data to my Adafruit IO feed.

Code for Adafruit Test:
```cpp
// This example uses an Arduino MKR GSM 1400 board
// to securely connect to shiftr.io.
//
// IMPORTANT: This example uses the new MKRGSM library.
//
// You can check on your device after a successful
// connection here: https://shiftr.io/try.
//
// by Sandeep Mistry
// https://github.com/256dpi/arduino-mqtt

#include <MKRGSM.h>
#include <MQTT.h>

const char pin[]      = "";
const char apn[]      = "internet";
const char login[]    = "";
const char password[] = "";

GSMClient net;
GPRS gprs;
GSM gsmAccess;
MQTTClient client;

unsigned long lastMillis = 0;

String hostName = "io.adafruit.com";
int pingResult;

void connect() {
  // connection state
  bool connected = false;

  Serial.print("connecting to cellular network ...");

  // After starting the modem with gsmAccess.begin()
  // attach to the GPRS network with the APN, login and password
  while (!connected) {
    if ((gsmAccess.begin() == GSM_READY) &&
        (gprs.attachGPRS(apn, login, password) == GPRS_READY)) {
      connected = true;
    } else {
      Serial.print(".");
      delay(1000);
    }
  }

  Serial.print("Pinging ");
  Serial.print(hostName);
  Serial.print(": ");
  
  pingResult = gprs.ping(hostName);
  
  if (pingResult >= 0) {
    Serial.print("SUCCESS! RTT = ");
    Serial.print(pingResult);
    Serial.println(" ms");
  } else {
    Serial.print("FAILED! Error code: ");
    Serial.println(pingResult);
  }

  Serial.print("\nconnecting...");
  while (!client.connect("6aa87eeb0623497ba86a756c0972f961", "gamelaster", "c4ddc1a29e444805aab4d0cb22a89037")) {
    Serial.print("last error: ");
    Serial.print(client.lastError());
    Serial.print("return code: ");
    Serial.print(client.returnCode());
    Serial.println(".");
    delay(1000);
  }

  Serial.println("\nconnected!");

  // client.subscribe("/hello");
  // client.unsubscribe("/hello");
}

void messageReceived(String &topic, String &payload) {
  Serial.println("incoming: " + topic + " - " + payload);
}

void setup() {
  randomSeed(analogRead(0));
  Serial.begin(115200);

  // Note: Local domain names (e.g. "Computer.local" on OSX) are not supported by Arduino.
  // You need to set the IP address directly.
  //
  // MQTT brokers usually use port 8883 for secure connections.
  client.begin("io.adafruit.com", net);
  client.onMessage(messageReceived);

  Serial.println("Welcoem!");

  delay(1000);
  
  connect();
}

void loop() {
  client.loop();

  //if (!client.connected()) {
  //  connect();
  //}

  // publish a message roughly every second.
  if (millis() - lastMillis > 1000) {
    lastMillis = millis();
    client.publish("xxxxx/feeds/xxxxx", String(random(2500, 3000) / 100));
  }
}
```

Any ideas?

Thanks

momor10, 2018-12-04:
i have the same issue. after updating the MKRGSM Library from 1.1.2 to 1.3.1 (using the MQTT Library version 2.4.1) i get error code -6 (LWMQTT_NETWORK_FAILED_WRITE).
Downgrading MKRGSM back to 1.1.2 works. 
So there seems to be some incompatibility caused by an MKRGSM update

gamelaster, 2018-12-17:
I was working with the things last days, everything was working well (I was not using connected function), I will try to get my SWD debugger working and I will try to find the problem

fnabarlats, 2019-05-24:
Hi,  I have the same problem. Actually I'm using MKRGSM 1.4.1 and mqtt 2.4.3
Any solution for this? Thanks

256dpi, 2020-10-18:
Closing as outdated. Please reopen if problem persists.

closed 2020-10-18 by 256dpi