256dpi/arduino-mqtt #132: feature request

burner-, 2018-11-02:
Currently there is only onMessage(messageReceived) callback. It would be nice that user can make routed subscribes that whay:
client.subscribe("/hello",myHelloHandler);   and that mqtt client lib will automatically route all /hello messages to that myHelloHandler. To save memory it would be even better if library supports wildcards so user can do:
client.subscribe("/hello/%" ,myHelloHandler);
or
client.subscribe("/hello/%/byebye" ,myHelloHandler);
and that will cause that it will get routed to myHelloHandler(int handlerid, String message )

Urs-Eppenberger, 2018-12-17:
I support this idea. It would make for cleaner, better readably and maintainable code, at least in my case.

I have a code for devices which subscribe to two topics, one for commands and one for measurement values from other sensors. In the callback routine, I currently need to check first the topic and then decide what action to take based on the topic. With the above feature, I can split the code in two callback routines, with clearly dedicated tasks.

I'm not very fond of the wildcard idea, though. The library is used primarily in small devices, where the need for wildcarding is rare. But I can see the one or the other use case for this too.

abcd-ca, 2019-06-08:
@256dpi what is your strategy when trying to parse which topic it is within your if/elseif/else when the topic has a wildcard like `foo/+/bar`? You can't just use String.equals() because the topic `foo/123/bar` obviously won't match `foo/+/bar`. Do you use a regex lib?

256dpi, 2019-06-09:
You can use `s.startsWith("foo/")` to match the beginning, then use `s.indexOf("/", 4)` to get the offset of the next "/", then get the segment in the middle using `s.substring(4, offset)` and use the same approach to get the end of the string using `s.substring(offset+1)`.

abcd-ca, 2019-06-10:
@256dpi good call, thanks.