256dpi/arduino-mqtt #134: mqtt over websockets?

debsahu, 2018-12-10:
Great work here Joel, I was wondering how difficult would this be to code this mqtt library over websockets. I'm asking because AWS IoT only supports MQTT over websockets.

256dpi, 2018-12-13:
Good idea! Since `lwmqtt` focuses only on the MQTT protocol, we should implement this in this project. Possibly we could use the following library: https://github.com/Links2004/arduinoWebSockets.

debsahu, 2018-12-13:
Yes agree, I have a [not so elegant implementation](https://github.com/debsahu/esp8266-arduino-aws-iot-ws) with a wrapper for Arduino IDE from [source found here](https://github.com/joekickass/esp8266-arduino-aws-iot-ws). Thinking of complete rewrite as now NTP is completely integrated with ESP8266 core.

debsahu, 2019-01-17:
@256dpi After the recent commits to ESP8266 regarding BearSSL, I was able to connect to AWS using regular MQTT protocol without the need for websockets. https://github.com/debsahu/ESP-MQTT-AWS-IoT-Core I guess there is no need for spending time on something that already works!

closed 2019-01-17 by debsahu